RavenSystem/esp-homekit-devices #302: T3 US touch turns on by its self

xtremedays, 2019-03-23:
Hi all,

have 3 T1 us switches, 3 channel ones all set up as required, (device 11, board type 1) and all three switches turn on at random. Using latest firmware. No log as using on 220v.... 

any thoughts?

xtremedays, 2019-03-24:
Tried using one as a 4ch switch (mode 4), still does the same thing.

RavenSystem, 2019-03-26:
It is an AC interference. Check wire connections and ensure that all wires are well connected.

xtremedays, 2019-03-28:
Yup wires are check and all ok. I actually suspect its something to do with stray RF turning them on. After disconnecting the pins that power the RF chip it doesn't seem to do it.. is that something that can be done via software? to ignore the RF inputs?

xtremedays, 2019-03-31:
Still having issues with the units. RF Chip removed. Have wired to 240v with no output connected and still turns on at random by themselves.

xtremedays, 2019-04-18:
Looks like a hardware problem with the 3 channel touch... too sensitive so have to change capacitors on the board.. still working out what it actually needs...

lizzus, 2019-04-29:
I have same problem with a NodeMCU connected with 3 relay

RavenSystem, 2019-05-11:
@lizzus @xtremedays Try version 0.8.9.

RavenSystem, 2019-05-11:
Better with 0.8.10.

xtremedays, 2019-05-13:
Updated to 8.10, had a ghost switch overnight last night

xtremedays, 2019-05-16:
but hasn't done it since......

RavenSystem, 2019-05-16:
At least, it seems that has been improved.

ValentinusGVR, 2019-05-19:
Hi

I have 8pcs. Sonoff Touch v.0.8.10
All this switches have ghost effect)))))
Sometime it’s switch on by yourself in night, sometime in the during day.
One-Two of this Eight switches have ghost switch in 24 hours.

But....

First Sonoff Touch arrived 1-2 months ago. I’m flash by v.0.7.3 (Without OTA) - on this firmware I don’t remember same ghost effects.

Maybe it’s problem with sensor button on Touch?
Because I don’t have same problem with Basic, Dual, TH modules.

About debugging.....
I think need some mechanism for take debug information from working device. For example write all debug in log file and share by internal web server http://192.168.0.4/debug.log and see inside who send commands for switch on 🤔

RavenSystem, 2019-05-19:
Try version 0.8.12 and setup new option "Button filter" to 50 (reboot required), for example:

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#button-filter-default-0

ValentinusGVR, 2019-05-23:
In last 2 days I have 3 times by self on/off 
In my office twice light in on by self last two night.
In kitchen in morning today is off by self.
Problem still exists.
Firmware 0.8.12 with Button filter = 50

Note: I have 8 sonoff touch.

My idea is create a possible to get log by remote (without ttl) because it’s inpossible to connect all 8 switches by ttl and listen for long time.

xtremedays, 2019-05-23:
So far so good for me. 4 days and no ghosting

RavenSystem, 2019-05-23:
@xtremedays Excellent!!!

@ValentinusGVR Try setting Button filter to 100 (max value) and reboot. Getting log by remote doesn't make sense because you only will see that your device has turned on or off, but logs don't tell you the cause. You can try to reduce AC interferences from your electrical installation too.

ValentinusGVR, 2019-05-25:
In my Sonoff Touch problem still exist.
Last 2 days I have 3 times on/off by self with Button Filter:100

I have two way:
Try to use default sonoff firmware for test a same problem.
What you think?

RavenSystem, 2019-05-26:
You can try original Sonoff firmware, but I can not do anything more with RavenCore, sorry.

advintl, 2019-05-30:
I got the same problem with T1 US 3C in version 0.9.0
So I take out the PCB which three touch panels on it , then connect it to my UART monitor program.
When the ghost effect showup, I got those log.
```
!!! AdvButton: No function defined
RC > Toggle SW 3
RC > Relay 3 -> 0
RC > Saving last states
```
and this
```
RC > Checking factory default call
RC ! Reset to factory defaults not allowed after 60000 msecs since boot. Repower device and try again
RC > Toggle SW 1
RC > Relay 1 -> 0
RC > Saving last states
```
durning the test , I don't touch the PCB or switch sensor

ValentinusGVR, 2019-06-02:
Master - please comment previous post with log.
it’s normally?

I still have problem(((  and don’t test with sonoff firmware because cant find for sonoff touch.

ValentinusGVR, 2019-06-04:
I was configure Sonoff Touch as "Socket + Button" and don't defined buttons for 1,2 and long tap.
But my lights is "On" by self in night.

When I set "Socket + Button" is still on-off works for very long push.
- From wiki: "To use the normal function of the button to toggle socket on/off, you must use a long press of 3 seconds."

Button filter is kill short pushes by self its good, but how you see we are have a long pushes by self.

What heed to do for fix on/off by self:
For device type "Switch 1ch" need new Button filter with "low" and "high"

For example:
We are set "Button filter low":50 - its will filter short push less 50ms
And set "Button filter high":200 - its will filter long push over 200ms

@RavenSystem please help, my wife is make brain storm for me every day about light((( - if need you can make custom build for test (without OTA). I will flash all my 8 Sonoff Touch write to you about results.

xtremedays, 2019-06-11:
Seems to have done the trick for me.. tried to drop it to 25, had ghosting, tried 40, no ghosting for over a week. Closing issue.....

closed 2019-06-11 by xtremedays