RavenSystem/esp-homekit-devices #303: Implement momentary Switch for device type 1

i3laze, 2019-03-23:
Hi,
I'm thinking of modding a generic 5V PCB (for Winix Air Purifier) with Sonoff SV using man-in-the-middle approach :)

I want to highjack power-on push button (SW1) as GPIO14 input (will be External Toggle: 1), 
however I want RavenCore to provide a 1 sec pulse as an output, like it currently does for Garage Door (device type: 8).

The on/off state should better be memorized by software and support setting initial state On (1 pulse on startup). Any chance you'll add this?

P.S. It is currently possible with Tasmota, but I love RavenCore more :)
[image]
[image]

yukonx, 2019-03-23:
Type 13 is currently available

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#13-lock-mechanism

i3laze, 2019-03-23:
Itching relay, now I get it! :)
Will give a try.

i3laze, 2019-03-23:
Indeed type 13 works as I require. But... it represents a Lock!
And it doesn’t save state :(

I need a Fan/Switch!
odd click - On, even click - Off.

RavenSystem, 2019-04-12:
Version 0.8.2 will do it.

RavenSystem, 2019-04-13:
Supported in v0.8.2.

closed 2019-04-13 by RavenSystem