RavenSystem/esp-homekit-devices #305: Función RF sonoff habilitada

arturodiazad1, 2019-03-24:
Hola, he instalado varios sonoff touch dual como tipo de dispositivo 12, perciana, en los toldos de mi casa, va genial,  estoy muy agradecido por el proyecto. Sabéis si se queda habilitada o se puede habilitar la señal RF en el sonoff touch? Estoy pensando en crear un anemómetro con Arduino y programarlo para cuando el viento pase de cierta velocidad envié una señal RF al sonoff touch y active el canal para cerrar el toldo e incluso incluir un sensor lumínico y que active el canal para abrir el toldo o un sensor tipo FC-37 para cuando llueva bajar la perciana. 
Si veis que puede ser viable esta idea comentarla, ya que creo que puede ser muy útil.

RavenSystem, 2019-03-24:
Hola Arturo, yo no tengo ningún dispositivo RF, pero hasta donde yo sé, la parte RF va fuera del esp8266, con lo que es independiente del firmware que instales. Aquí ya he podido leer varios comentarios indicando que la RF funciona perfectamente después de instalar RavenCore, y nunca que visto ninguno alegando problemas al respecto.

arturodiazad1, 2019-03-24:
Si hacerlo a través de HomeKit sería lo mejor, pero he pensado en esta opción por que con mis pocos conocimientos me parece más asequible. 
Muchas gracias por tu respuesta, iré investigando.

closed 2019-04-08 by RavenSystem