RavenSystem/esp-homekit-devices #307: problem Sonoff 16TH

mattsharks, 2019-03-25:
hi,
I followed your sonoff tutorial. I flashed Sonoff TH16. relay and temperature are displayed on homekit and Eve. Everything works if I ask to turn on at  a certain time.
but if I set a tempereature setpoint, the heating never start and never stop.
have you ever tested this material?
Please Help !

RavenSystem, 2019-03-25:
I am using 3 Sonoff TH16 as thermostats without any issue for a year.

mattsharks, 2019-03-25:
Ok
I am using this thermostat ( Ds18b20)
https://www.amazon.fr/dp/B07KT2N828/ref=pe_3044141_185740131_TE_item
Th sensor type : 3 ( don’t work)

But with th sensor type 2 (work)

LyoshaDovolen, 2019-03-27:
Same case here (TH16, Sensor Type 3, 0.7.3) and have "no response" after a few minutes of work.

lizzus, 2019-03-28:
Any bug on 0.7.3 for sensor?

mattsharks, 2019-03-28:
I use the sensor type 3 (immersible) on the sonoff TH16 with version 0.7.3.
I can order either with the clock or live.
The temperature is well up.
The problem comes when I want it to be an instruction that drives the water heater. it does not work.
Heater <35 ° c
Stop> 50 ° c
whether the start or stop instruction, none works

j'utilise le capteur type 3 (immergeable) sur le sonoff TH16 avec la version 0.7.3.
j'arrive à commander soit avec l'horloge soit en direct.
La temperature est bien remonté.
Le probleme vient au moment ou je souhaite que ce soit une consigne qui pilote le chauffe eau. ça ne fonctionne pas.
Chauffe <35°c
Arret >50°c
que ce soit la consigne de démarrage ou d'arret, aucune ne fonctionne

closed 2019-04-16 by RavenSystem