RavenSystem/esp-homekit-devices #308: wemos d1 mini

sorriso93, 2019-03-25:
Hello
I successfully put the firmaware on a standard nodeMCU, but I can't get it running on wemos d1 mini board; no AP wifi is available.
This is the serial output with OTA.bin and main.bin (072 & 073)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.

pandarek, 2019-03-26:
I have a similar problem with Wemos D1 mini and OTA. I uploaded the version without OTA and it works fine.

ymyshko, 2019-03-26:
Wemos D1 mini lite, ESP01 with 1Mb flash run with OTA fine.
Wemos D1 mini with 4Mb flash not run with OTA. Only without.
It is my experience.

sorriso93, 2019-03-28:
> I have a similar problem with Wemos D1 mini and OTA. I uploaded the version without OTA and it works fine.

uups... you used this command?

`esptool.py -p /dev/cu.wchusbserial1430 --baud 115200 write_flash -fs 4MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin`

becouse I tried with both 072 and 073 with no success...

pandarek, 2019-03-28:
Yes, I'm using this command.
Uploaded by built-in D1 mini usb uart converter. I use these drivers:
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

I was able to download and install firmware via OTA, but after restarting I need to reconfigure wifi to add accessory in HomeKit. And then in settings (EVE) there is no option Firmware Update.

sorriso93, 2019-03-31:
Solved, wemos d1 mini needs D3-GPIO0 to ground before flashing!

closed 2019-03-31 by sorriso93

pandarek, 2019-04-01:
This is obvious ;)
However, there are no solves problems with OTA for Wemos D1 mini.