RavenSystem/esp-homekit-devices #309: Add API Interface

Tomcuzz, 2019-03-25:
This project is amazing currently! Thanks for the awesome work!
I would like to be able to run scripts from a server to interface with this,
Would it be possible to open up a web api, a socket api or mqtt api (Just some thing I can connect to from a python script) to allow control of the state and return back the current status?

mmaton, 2019-03-25:
Sounds like a good idea, I know you want to aim for upmost security but sometimes we do need to control devices via other means. API_TOKEN and API_ENABLED could be a device setting. Not sure how certificates might be handled.

RavenSystem, 2019-03-26:
Sorry, only HomeKit will be available for security matters.

peros550, 2019-03-26:
@RavenSystem 

you have build an amazingly **stable** firmware. You have been dedicated to this project the last months fixing bugs and adding new features and devices. We can't help you enough for what you have achieved! 

In regards to this topic's request, I fully support the idea. There is a balance between security and usability. If you lean towards security you have to compromise the usability and vice versa. 

Why not give this option to the user though? Security can still be retained if LAN environment is secure enough.  Somebody who wants to use more features would be able to secure better their LAN and choose to open these protocols (i.e. restAPI, Http or mqtt). 

Perhaps it could be added to your new Home Architect firmware as an option? 

I can see great potential if you provide this functionality. There are third party Home Management environments (like openHab, HomeAssistant, etc)  that provide more perks features and devices than what Apple Homekit support. Integrating Sonoffs, Wemos etc using your stable firmware would be amazing. 

Of course, we should consider that all these special requests mean "time" and you are already busy! 
Thanks for reading so far and I really think you can consider it.

mmaton, 2019-03-26:
@Tomcuzz Would you be keen to work on a PR or a fork for this? I'd happily help you

Tomcuzz, 2019-03-26:
@peros550 Thanks for your comment, couldn't agree more with what you have said.
@mmaton Yep think that may be the way forward if @RavenSystem Doesn't feel its a possible feature/doesn't have the time to implement

Some background on my reasoning behind this, is I live in a shared flat with someone who uses an android phone :( so I need a way to bridge out to another system like HomeAssistant for his control.

My current system consist of Tasmota, MQTT and HomeBridge but this would simplify the whole working.

RavenSystem, 2019-03-26:
Sorry, maybe my explanation may have been too short. HomeKit eats many resources, this is the reason because Tasmota does not include HK, because it can not do it keeping security and stability (https://github.com/arendst/Sonoff-Tasmota/issues/3876#issuecomment-426659265). Running HomeKit and others can produce a stack overflow, that is a security issue and a serious crash cause. And stability will be affected. Remember that esp8266 is a 80MHz 1 core CPU with about 64KB of RAM.

If you need HomeKit and others (rest API, MQTT...), I suggest to use Tasmota and HomeBridge instead.

Another way will be to use an esp32.

NorthernMan54, 2019-04-09:
Rather than adding another API, would it be feasible to just use the HomeKit API but without the security layer?  Similar to how Insecure Mode works with Homebridge?  If this was feasible, then my homebridge-alexa plugin would see these devices and Alexa enable them, and my other piece of work to enable NodeRED to control homebridge accessories would work with these as well.

closed 2019-05-10 by RavenSystem