RavenSystem/esp-homekit-devices #312: DHT11 without swith

cadetandrey, 2019-03-28:
Hi
How about new device type temperature Th sensor only ?
Thank you

lizzus, 2019-03-28:
Yes! I’d like it!!! I found this object with esp01s and ds18b20 sensor.

https://ibb.co/3Rbrs7k

cadetandrey, 2019-03-28:
Today add same sensor with dht11
set board 2 device 6 and have switch ))

lizzus, 2019-03-29:
This have not a relay! What do you do for have a switch?

cadetandrey, 2019-03-30:
Yes, have not relay, but in homekit I have switch unused

RavenSystem, 2019-03-30:
RavenCore does not support it. You must wait until HAA firmware is released.

lizzus, 2019-03-30:
Why not supper it?

pandarek, 2019-04-01:
@lizzus try this:
https://github.com/pandarek/esp-homekit-demo/tree/master/examples/temperature_sensor
If you can not do a compilation and set up a GPIO for the TH sensor, I can sort the bin.

lizzus, 2019-04-01:
@pandarek , please contact to my email lizzus[at]me.com thanks

RavenSystem, 2019-04-01:
> Why not supper it?

Because RavenCore is focused to work with Sonoff devices. HAA will support it, because it is focused in all kind of devices, including custom devices.

lizzus, 2019-04-01:
but inside ravencore esp01s is present. I use with device 10.  
Esp01 is economic and versatile... We can use for motion or themperature sensor...

RavenSystem, 2019-04-01:
@lizzus Yes, but I included esp01 for a personal custom project, and it was a bad idea. Because Esp01 is  versatile, there are many possible options to customize. RavenCore uses custom HomeKit characteristics to customize the device, and when a new custom HomeKit characteristics is created, performance lows a bit. Actually, I keep performance in a very good level, but I can not add the necessary characteristics to create a enough versatile custom device. That is the reason because I am developing HAA firmware: you will provide a json string to configure device instead using custom HomeKit characteristics.

closed 2019-04-12 by RavenSystem