RavenSystem/esp-homekit-devices #313: Sonoff Dual R2

matousek35, 2019-03-29:
Hello, at first I want to thank you for this amazing job!
My problem is that I bought Sonoff Dual R2 and it seems that it is not supported by the RavenCore. I tried to configure all types of boards (1-4) and I am unable to control relay 2. Is it fault on my side or is it just not supported? Thank you!

RavenSystem, 2019-03-29:
It is supported. What is your issue with details? Are you using device type 2?

matousek35, 2019-03-29:
I solved it. There was a badly soldered pin on one relay. It is working perfectly right now. 

But I have another question. I want to use that sonoff board for controlling my window blinds, but after completing closing cycle I need to reverse the direction of the motor but have no idea how to do that. I am more mechanical than electrical guy so I probably need little help in this area. Is it possible change the direction? Any extra circuit needed? 
Thank you! 🙂

RavenSystem, 2019-03-29:
To control a window covering, you must use Device type 12. Your motor must have 3 wires: N, L-up and L-down. N is neutral and is always connected, and Ls are connected to each relay to up and down (An earth wire can be present too, but it is for earth connection).

RavenSystem, 2019-03-29:
@matousek35 Are you sure that your motor is DC and not AC?

matousek35, 2019-03-29:
Yeah, sure. I want to use it on 5V not 230V. I solve it with external relay board (with 2 relays) where are three inputs (COM, NO, NC). These two extra relays are driven from Sonoff outputs. I know that it is not the best solution but it works. I want to make a covering system for roof windows based on power from small solar panel.

yukonx, 2019-03-29:
use dc power 
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#1-switch-1ch

sonoff 
type 1 
 2 realy

https://drive.google.com/open?id=1ECxFD5EaXhlMmAgdzugKSBoanLs49N-_

ThierryDB, 2019-04-02:
Hi, Just installed RavenCore on 2 sonoff basic and 1 sonoff dual R2. When adding then to Apple Homekit no issues the Basic but the Dual shows as one Switch and I can't see how to turn on relay 1 or 2 individually. Is that function supported? thanks for the great works.

ThierryDB, 2019-04-02:
Hi one more thing. Before flashing my sonoff dual I could turn on the two lights that I have wired to my sonoff dual individually with the original app. Now only relay 1 turns on? Is that a bug?

RavenSystem, 2019-04-02:
Read Wiki, please.

closed 2019-04-12 by RavenSystem