RavenSystem/esp-homekit-devices #315: Sonoff 4ch

STijeras91, 2019-03-30:
Buenas, ante todo gracias por el trabajazo, me gustaría saber si con el sonoff 4ch se podrían activar los external toggles, para utilizar interruptores de casa normal, a parte de poder controlarlo con HomeKit. gracias!

skop1986, 2019-04-01:
STijeras91 
Physical buttons on sonoff ch4 pro work!

lfiocco, 2019-04-01:
When you press the buttons on the unit, homekit also refreshed this information?

Have you paired it with a remote. How does it work? Can you toggle the remote buttons and see what’s happening on the homekit app?

STijeras91, 2019-04-01:
> When you press the buttons on the unit, homekit also refreshed this information?
> 
> Have you paired it with a remote. How does it work? Can you toggle the remote buttons and see what’s happening on the homekit app?

Yes, the buttons integrated in the Sonoff 4CH board work perfectly and update the Homekit status.

STijeras91, 2019-04-01:
> Physical buttons on sonoff ch4 pro work!

With the non-PRO version, can it also work?

lfiocco, 2019-04-06:
I can confirm that on the pro R2 the remote and local buttons work, after a few seconds the changes are seen in the home app.

closed 2019-04-16 by RavenSystem