RavenSystem/esp-homekit-devices #316: ISSUE - Water valve : automation Homekit

claire078, 2019-03-30:


lfiocco, 2019-03-30:
I think this is an Apple limitation. You can using a different app to create the automation. I think Apple wanted manufactures to create schedules for watering.

RavenSystem, 2019-03-31:
@lfiocco you are right.

claire078, 2019-03-31:
😏
In this case, there is no make sense to set to water valve. Simple switch is the best way to make automation. We just lose the icon in Home app.

closed 2019-03-31 by claire078

RavenSystem, 2019-03-31:
Water valve is a switch with a countdown, nothing more.

lfiocco, 2019-03-31:
I would not use a switch for controlling water flow. Water valves must have a safety measure in place, like a timer to switch off automatic and prevent flooding.

claire078, 2019-04-01:
> I would not use a switch for controlling water flow. Water valves must have a safety measure in place, like a timer to switch off automatic and prevent flooding.

That's right and I can agree with it. Other hand, by a watering system would be nice if we could timer set. That is the point, I don't want to start watering manual.

lfiocco, 2019-04-01:
If you have 6 zones or less I would get the Sonoff basic, isolate the relay and config it. That’s my solutions for now, I have given up on raspberry pi with Homebridge. I can’t keep it alive for more than 2 weeks. RavenCore and Sonoff are my new friends

RavenSystem, 2019-04-01:
@claire078 Water valve accessory has not a timer to start, it only has a working timer. Those are Apple defined characteristics. if you want to start in a specified hour, you must use an automation. If you can not do with stock Home App, you can try to do it with EVE App, or another.