RavenSystem/esp-homekit-devices #318: strange behaviour of garage door

pilot1981, 2019-04-01:
Hi,
I have a strange problem with garage door opener: sometime when I send open command, relay start and garage door receive input (I see light start), but after 2 seconds the light shut off and garage door don't open....then I have to send again open command and all works fine!

I already tried to upgrade from 0.6.2 to 0.7.3, but problem remain; so I tried to increase inching time from 0.7 s to 1.2 s, but this don't solve problem...

You have some suggestion for me?

regards

pilot1981, 2019-04-08:
it happen also with manual activation: I suppose this could be a problem of my garage door automation...you can close

closed 2019-04-08 by RavenSystem

pilot1981, 2019-04-14:
I continue to have issue. I have STARTS3 garage door opener. I notice problem happen always after many hours I don’t use it (about 8 hours).
So I ask: in SONOFF BASIC there are some types of inducted currents that could create this issue?

jat80, 2019-05-13:
I have a similar issue but it works fine with manual activation via my own garage door remote.

https://github.com/RavenSystem/esp-homekit-devices/issues/383

greggsymington, 2019-05-14:
+1
This has been plaguing me with one of my two doors. 

"Left" door works great, "Right" door will start doing it and even after deleting, reflashing and re-add to Home it lingers. Please see video [HERE](https://youtu.be/OTDG9QGhb3c).

.