RavenSystem/esp-homekit-devices #320: Tuya SWA1 with esp8266 Smart plug

massimo79, 2019-04-02:
Hy! I correctly flashed the Tuya Smart plug with the ravencore firmware since the esp is compatible. Only thing is that the GPIO connectivity is different from Sonoff and relay doesn’t trigger. GPIO should be something like this:
GPIO4: Led
GPIO5: Relay
GPIO13: Button
Since these plugs are very popular, could you make a “board type” for these?
Thanks from your friends in Italy!

GPL71, 2019-04-02:
@RavenSystem  Josè  it would be interesting a solution like the follow
The GPIO list where default set is NONE 
The user may choose and assigne the function to GPIO e.g. output, input, led and other....

What do you think?

RavenSystem, 2019-04-02:
@GPL71 HAA firmware will work in a similar way. You will can setup each GPIO using a JSON string like:
{
   type: "switch",
   relay_gpio: 4,
   button_gpio: 8
}

massimo79, 2019-04-02:
Will the setup be available also in Eve?

GPL71, 2019-04-02:
@RavenSystem Josè without words .... Great!!!!
Have you any idea when we'll see a first release about?
@massimo79 I think by web ... Josè wrote something about ...
I think this firmware will be something we can't image at the moment....

massimo79, 2019-04-02:
@RavenSystem I can beta test (if you like) the GPIO settings in the new HAA when the beta is available.

jorgemartins, 2019-05-05:
Hey @massimo79, out of curiosity, have you flashed the tuya device through the serial port or with OTA tool like Tuya-convert ?

I'm asking because @RavenSystem just recently created a joined `.bin` #364 that could be used in a OTA approach, but you flashed it before him doing that so you could have your methods :)

massimo79, 2019-05-05:
Hi @jorgemartins, I flashed the Tuya switch with a usb serial adaptor and no need of soldering. Hard erasing and uploading new firmware had no issues, problem is the GPIO settings that don’t make it work!

jorgemartins, 2019-05-14:
hey @massimo79 just noticed HAA was released → https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect

have you tried it to use the correct GPIO? 🤞

massimo79, 2019-05-15:
Hi @jorgemartins not yet! Planning to do so this weekend and I can’t wait to try. I’ll give you feedback if I succeed and will share the json!

RavenSystem, 2019-05-16:
JSON for HAA: `{"c":{"l":4,"i":1},"a":[{"t":2,"r":[{"g":5}],"b":[{"g":13,"p":1,"t":1}]}]}`

massimo79, 2019-05-17:
Great @jorgemartins thanks a lot! Con you confirm that I have to reflash the device? (Right now it’s running ravencore).

jorgemartins, 2019-05-17:
I think you meant José and not Jorge 😅.
Anyway credit goes to @RavenSystem - thank you for your work, by the way!

Regarding your question I think you need to reflash as HAA is a different firmware. I'm not sure if we can simply change the OTA repository without reflash. Can we @RavenSystem ?

massimo79, 2019-05-17:
Sorry @RavenSystem..my mistake! 🤪

jorgemartins, 2019-05-20:
Hey @massimo79, you closed the issue but didn’t mention if you were successful with having homekit on your tuya device 😬.
Thank you

RavenSystem, 2019-05-20:
To install HAA is necessary to erase flash and start from scratch.

massimo79, 2019-05-24:
@jorgemartins unfortunately no! The device doesn’t work correctly in AP mode (the first time LCM showed up but wasn’t able to add device to Homekit, in other attempts LCM never showed up and I wasn’t able to download HAA). I will try again!

massimo79, 2019-05-24:
@jorgemartins @RavenSystem issue closed by mistake and reopened.

RavenSystem, 2019-05-29:
I suggest to flash LCM firmware, and then connect device to 230VAC and do the rest of installation (wifi setup, repo, ...).

franklindee, 2019-06-05:
hi do you know how to reset the HAA and redo the configuration?

closed 2019-07-20 by RavenSystem

mientki, 2019-08-23:
> JSON for HAA: `{"c":{"l":4,"i":1},"a":[{"t":2,"r":[{"g":5}],"b":[{"g":13,"p":1,"t":1}]}]}`
> 
> https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#tuya-swa1-with-esp8266-smart-plug

Will it work with the Tuya PP-W162 socket? How can I change the software without soldering the wires. Access to the pins is so difficult and requires desoldering the entire motherboard. Hour of work.