RavenSystem/esp-homekit-devices #321: Water valve

netRunner0, 2019-04-04:
Hey. You use the device type 7 - Water Valve in your firmware. However, it looks like an irrigator, as it has a timer to shut off, which will not allow it to be used as a water control system.

Can you add this service as a tap? the tap has no timer and that would be very cool

netRunner0, 2019-04-04:
[image]
[image]

rechin304, 2019-04-04:
Check Wiki page

7. Water Valve

Use your device as a HomeKit Water Valve or Irrigation System, with a built-in timer.

Valve Type [Default: 0]

Reboot required when changed.

Select the water valve type icon to show:

0. Water Valve
1. Sprinkler
2. Shower
3. Tap

So use valve type 3

netRunner0, 2019-04-04:
With this configuration, the situation does not change, there is still a timer

netRunner0, 2019-04-08:
[image]

RavenSystem, 2019-04-08:
Actually, there is no way to remove timer. A future version could bring that option.

RavenSystem, 2019-04-09:
Supported in RavenCore v0.8.0.

closed 2019-04-09 by RavenSystem

rechin304, 2019-04-09:
After update to v0.8.0 and enable Have Stop, where do you setup the time ?

RavenSystem, 2019-04-09:
time no, timer. The new feature is the option to disable the timer.

rechin304, 2019-04-09:
Sorry, I wanted to say timer. As now the timer is 15 sec then is off.

RavenSystem, 2019-04-09:
> Sorry, I wanted to say timer. As now the timer is 15 sec then is off.

What do you mean?

rechin304, 2019-04-09:
Before was default duration

RavenSystem, 2019-04-09:
Timer is a HomeKit characteristic. Nothing has changed with water valve timer working way.

rechin304, 2019-04-09:
That option is not anymore in Home app.

RavenSystem, 2019-04-09:
In my case works like a charm. Remember to reboot your accessory when "Has Stop" option is changed.

rechin304, 2019-04-09:
ok, reboot was the issue. thanks. Maybe is good idea in wiki to mention that a reboot is required.

RavenSystem, 2019-04-09:
It is in the Wiki.

rechin304, 2019-04-09:
Ok, then probably I missed. Thank you for your efforts !!

netRunner0, 2019-04-09:
[image]

RavenSystem, 2019-04-09:
Don’t worry. Many thanks!!