RavenSystem/esp-homekit-devices #323: Problem with sensor close on garage door

pilot1981, 2019-04-07:
Hi,
I have sonoff basic new version R2 with cutter track and ravencore 0.7.3 using as device type garage door (I have another garage door working fine): all worked fine but I had problem with my close sensor for about 3 weeks so garage door result always opened for this time. 
Now I repaired my sensor close but eve app and Apple home app don’t receive change of state of sensor close....

I already tried to do software and hardware reboot, WiFi reset and removed/added accessory to Apple home app...without any result....

Before flash again, what you suggest to do? There is another way to reset all?

lizzus, 2019-04-08:
Maybe problem is sensor...

pilot1981, 2019-04-08:
problem non close sensor excluded.

I done more troubleshooting and discovered the issue on close sensor happen only using 220V; when I use 3.3V supply (powered by USB of macbook) all work fine!

I captured also these logs:

#!A߬!Cܟ!Aoq9-Bc)!))4add 0
aid 6
cnt
connected with Vodafone-A76824481, channel 11
dhcp client start...
ip:192.168.1.110,mask:255.255.255.0,gw:192.168.1.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
RC > Creating accessory name
RC > Creating HK accessory
RC > Creating Setup accessory
RC > Starting HomeKit Server
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 9A:CE:1B:E3:64:D0
>>> HomeKit: Found admin pairing with 226B3592-2307-44C6-9C4D-E087492C6239, disabling pair setup
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore-924F48 md=RavenCorepv=1.0id=9A:CE:1B:E3:64:D0c#=451s#=1ff=0sf=0ci=4 Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Get Accessories
RC > Returning target_door_state -> 1
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
RC > Returning target_door_state -> 1
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
RC > Returning target_door_state -> 1
>>> HomeKit: [Client 4] Update Characteristics
RC > GD activated: Current state -> 1, Target state -> 0
RC > GD relay working
RC > GD -> OPENING
>>> HomeKit: [Client 5] Update Characteristics
RC > GD -> OPENED
RC > GD -> CLOSED
RC > GD -> OPENING
RC > GD -> CLOSED
>>> HomeKit: [Client 4] Update Characteristics
RC > GD activated: Current state -> 1, Target state -> 0
RC > GD relay working
RC > GD -> OPENING
RC > GD -> OPENED
RC > GD -> CLOSED

but in my opinion log sound s good...so why it work fine with 3,3v but don0t work fine with 220V?

@RavenSystem can you help me?

regards

pilot1981, 2019-04-08:
I already flashed again with 0.7.3 but this don’t solved!

Another detail: close sensor is attached using a cable with 10 meter of length...initially all worked fine...

lizzus, 2019-04-08:
10m is too much

pilot1981, 2019-04-08:
@lizzus You know the maximum length for GPIO cable?

lizzus, 2019-04-08:
What AWG cable you use?

pilot1981, 2019-04-08:
Wrong behavior happen with cable 5 cm...

RavenSystem, 2019-04-08:
Are you sure that sensor works fine? Try to use a simple wire to join gpio and gnd. For sure it is a hardware issue.

pilot1981, 2019-04-09:
Yes, I already tried with cable to join RX to GND: the issue happen only with 230v...

It’s possibile read console log when 230v connected?

pilot1981, 2019-04-09:
Another detail:
- with 220v: relay work fine, but close sensor RX don’t work
- with 3.3V: relay don’t work, but close sensor RX work fine

any idea?

pilot1981, 2019-04-09:
I suppose this is normal. "with 3.3V: relay don’t work, but close sensor RX work fine"

I tried to erase again flash and install new version v0.8.0, but with 220v I have alway same problem: relay work fine, but close sensor RX don’t work(using sensor cable with length of 5 cm)...

any idea?

SONOFF damaged?

pilot1981, 2019-04-10:
Hi,
these are photos of my sonoff and config:
[image]
[image]
[image]
[image]

RavenSystem, 2019-04-10:
When your door is closed, does your sensor join rx pin and gnd?

pilot1981, 2019-04-10:
yes, contact colse correctly but in apple home app remain in opening state...

RavenSystem, 2019-04-10:
What happens if you remove sensor and simply join the wires?

pilot1981, 2019-04-10:
if I connect GND and RX pin using a simple cable? same result: SONOFF don't understand open/close signal...

RavenSystem, 2019-04-10:
And same test with Sonoff connected to 3.3V?

pilot1981, 2019-04-10:
with 3.3v GPIO work fine

RavenSystem, 2019-04-10:
IMO it is a hardware issue of your Sonoff.

pilot1981, 2019-04-10:
you know howto troubleshoonintg with 220V?
attach 220v to sonoff and then attach sonoff to usb using FTDI without VCC and GROUND PIN?

pilot1981, 2019-04-10:
You know why two sonoff with ravencore 0.8.0 have different states in Apple home app during opening stage?

[image]

pilot1981, 2019-04-10:
You see my photo?

RavenSystem, 2019-04-10:
> You see my photo?

No.

pilot1981, 2019-04-10:
[image]

pilot1981, 2019-04-10:
Why one sonoff say opened and the other one say opening (the one with problem on close sensor)? Both were in opening state!
Both with 0.8.0 version

RavenSystem, 2019-04-10:
Maybe different sensor types and EVE setup is wrong in one of them.

RavenSystem, 2019-04-10:
Or maybe one of them has a hardware issue.

pilot1981, 2019-04-10:
Both have close sensor NC and same EVE config: this confirm HW damage...

Please, you can suggest exact instructions to capture logs during 220v power supply? I don’t want destroy my MacBook... ;-)

RavenSystem, 2019-04-10:
That is a risk.

A better way first: change device type to 1, and set external toggle = 2. The function that controls garage door sensor is the same that manages external toggle. So, you will be able to check the RX-GND trigger directly in your Home App.

(Remember that relay doesn't work with 3.3VDC, because relay needs 5VDC).

pilot1981, 2019-04-12:
ok, I go to try it with device type=1 and board type=4, but what I have to check? I have to use electrical tester on RX-GND contacts when I go to switch using apple home app? I understood well your suggestion?

pilot1981, 2019-04-12:
or I have to check opened/closed in apple home app when I switch on/off?

RavenSystem, 2019-04-12:
> I have to use electrical tester on RX-GND contacts when I go to switch using apple home app?

No.

> or I have to check opened/closed in apple home app when I switch on/off?

Yes. You must use RX-GND as external switch. (A garage door sensor is like an external switch).

pilot1981, 2019-04-12:
ok tested. I encountered same issue: with 220V RX-GND don't work...very strange it works fine with 3,3V....I go to buy a new SONOFF...

RavenSystem, 2019-04-12:
@pilot1981 Definitely, it is a hardware issue of Sonoff device.

pilot1981, 2019-04-12:
thanks for support!

closed 2019-04-12 by RavenSystem