RavenSystem/esp-homekit-devices #324: Revencore como pulsador

PinchXo, 2019-04-08:
querria poner un shelly 1 o un sonoff con reverncore para pasar un hilo de maniobra sin tension para que desde home kit dando un toque se cierre el contacto durante 1 o 2 segundo y se quede en ON y dandole otro toque quede en OFF  seria una señal de mi aire acondicionado con un toque se enciende y con otro toque se apagaria que configuracion tendria que tener en Revencore

gracias 

I would like to put a shelly 1 or a sonoff with reverncore to pass a maneuver thread without tension so that from home kit giving a touch to close the contact for 1 or 2 seconds and stay ON and giving it another touch to be OFF would be a signal of my air conditioner with a touch turns on and with another touch it would turn off which configuration would have to have in Revencore

Thank you

RavenSystem, 2019-04-12:
https://github.com/RavenSystem/esp-homekit-devices/issues/303

closed 2019-04-12 by RavenSystem

RavenSystem, 2019-04-13:
Supported in v0.8.2.

PinchXo, 2019-04-13:
ya esta v0-8.2??  se puede actualizar ya?