RavenSystem/esp-homekit-devices #326: magic home bug

curla92, 2019-04-09:
Hi josè. i have just tried your new update for the magichome. I managed to fix the colors but i can't switch off the magichome.. when i tried to turn off the strip goes to red color...
Can you help me? thank y

RavenSystem, 2019-04-09:
Have you check RGBW pins setup?

curla92, 2019-04-09:
GPIO 5 12 AND 13.. GPIO 5 is the first pin after the voltage?

RavenSystem, 2019-04-09:
RGB GPIOs 5, 12 and 13 are the default setup, but your R pin will be other than 5. I don't know how is your board, but you can search tracks of esp8266 and check them here: https://github.com/esp8266/esp8266-wiki/wiki/Pin-definition

curla92, 2019-04-09:
i have magichome zj-wfmn-a v1.1

lizzus, 2019-04-09:
Magic Home GPIO https://ibb.co/n0HwzvX

curla92, 2019-04-09:
mine is different

RavenSystem, 2019-04-09:
There are a lot of different Magic Home compatible devices. That is the reason because a custom GPIOs setup is present.

curla92, 2019-04-09:
i know. im looking for setup of my model. 

[image]
[image]

RavenSystem, 2019-04-09:
Can you do a photograph with better light?

curla92, 2019-04-09:
[image]
[image]
[image]

lizzus, 2019-04-09:
@RavenSystem sorry for O.T. 
I writed you.

RavenSystem, 2019-04-09:
In the photo, I see that used GPIOs are 5, 12 and 13. Red pin is managed with GPIO 5. But I see a critical point very near of RXD. The critical point inside circle is the GPIO 5; try to clean it.

[image]

rechin304, 2019-04-09:
I have same version and is working fine. R-5, g-12, b-13

curla92, 2019-04-09:
ok maybe it's this point.. damn it's very small!

curla92, 2019-04-09:
> I have same version and is working fine. R-5, g-12, b-13

turn off works?

RavenSystem, 2019-04-09:
@curla92 firmware has been tested and works perfectly (turn off included). Try to clean a bit the critical point. You only must remove the soldering tin next to the tiny hole to avoid contact, because the hole has not resin.

RavenSystem, 2019-04-09:
> @RavenSystem sorry for O.T.
> I writed you.

I saw your emails, don't worry. But what's is "O.T."?

lizzus, 2019-04-09:
over thread

lizzus, 2019-04-10:
**Magic Home Pin configuration by model:**

1. MAGICHOME LED CONTROLLER
PIN       14      // RED
PIN       5       // GREEN
PIN       12      // BLUE
PIN       13      // WHITE

2. MAGICHOME LED CONTROLLER 20
PIN       5       // RED
PIN       12      // GREEN
PIN       13      // BLUE
PIN       15      // WHITE

3. MAGICHOME ZJ WFMN A 11
PIN       12      // RED
PIN       5       // GREEN
PIN       13      // BLUE
PIN       14      // WHITE

4. MAGICHOME ZJ WFMN B 11
PIN       14      // RED
PIN       5       // GREEN
PIN       12      // BLUE
PIN       13      // WHITE

curla92, 2019-04-10:
hi, i want to know that i changed the maagic home device and this new works perfectly. thanks for the help josè, gpio5 on the last magichome was broken.

closed 2019-04-10 by RavenSystem