RavenSystem/esp-homekit-devices #327: Version 0.8 Thermostat -Deadband working?

lfiocco, 2019-04-10:
Updated to 0.8. Home reports 0.8. Testing dead band setting at 2.0. Set point at 23.0. Noticing that switch goes off at 23.1 and back on at 22.9. Testing with heat lamp on 18b20. Th16.

Anyone else getting this results?

RavenSystem, 2019-04-10:
There is a bug. I swapped + and - in deadband var.

RavenSystem, 2019-04-10:
Please, try 0.8.1 and comment.

lfiocco, 2019-04-10:
Set point 22.5
Deadband = 3.0
Unit turns off at 21.0
Unit turns on at 22.6
Looks like the upper limit is not calculated should be turning off 24.0 ?

I am using heat, not auto should I be using auto? 

AUTO will give you +/- 3.0
Heat will give you - 1.5
Cool will give you + 1.5 if so this is good. Really good ..

I will test tonight, I need the fan. Heheh

lfiocco, 2019-04-10:
No forgot AUTO = off

closed 2019-04-10 by RavenSystem