RavenSystem/esp-homekit-devices #328: My SSID is veeeery slow at reboot....

kalety, 2019-04-10:
Hi, Jose, first, I thank you, in my name, and the about 1000 persons belonging to spanish telegram group that you know, for you great, great and amazing job (incredible).
When I have to cut the light or a light cut happens, my ONT is extremely slow to reboot (near 3-5 min). All the devices reboots and wait the wifi SSID it is available. When this occurs, all the devices are connected to my wifi, except my (four) Ravencore, that remains "unavailables" and in the wifi list they appears with RavencoreXXXXXX, that is in AP mode. Can you enlarge the time that this devices are "hearing" and waiting that SSID is available?. Thank you again!

RavenSystem, 2019-04-10:
@kalety Many thanks!!

Sure, I can set delay time to 300 secs before device enters in AP mode. Will it be right?

RavenSystem, 2019-04-10:
Please, try 0.8.1 and comment.

kalety, 2019-04-10:
Thank you Jose, for answer so quickly! I think that five minutes will be enough... I will test it when I arrive at home!. Thanks again!

i3laze, 2019-04-10:
Actually, Tasmota has explicit wificonfig option for this case:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#Wi-Fi

I personally prefer “wait for the SSID forever”. Maybe there’ll be such an option someday?

RavenSystem, 2019-04-10:
> I personally prefer “wait for the SSID forever”

Me too.

> Maybe there’ll be such a choice someday?

Sure.

kalety, 2019-04-10:
It works flawlessly!!. Thank you again!

closed 2019-04-10 by RavenSystem