RavenSystem/esp-homekit-devices #329: Window covering - tilt angle

pixidixi88, 2019-04-11:
Hi,
Is it possible add to tilt angle steering? I use Sonoff Dual to operate roller shutters

RavenSystem, 2019-04-11:
Additional hardware will be necessary.

pixidixi88, 2019-04-11:
I have also this device https://blebox.eu/en/products/8-shutterbox-roller-shutter-controller-for-smartphone.html It is not supported by Homekit this is whyI would like to change to your super truper perfect solution :D BTW big thank you for all your efforts.

In this driver angle tilt control is made base on time of motor operating.
In the roller shutter is only one motor to operate both: rolling and tilt angle. If you roll down shutter then tilt is 90 degrees. when roll up -90degres. so to get 0 (open shutters - horizontal) you need to back ~1s in opposite direction.

I don't know if you can understand me but I can make short movie to show how it works.

pixidixi88, 2019-04-15:
Any idea when can we expect this feature?

RavenSystem, 2019-04-15:
No idea, sorry.

maxnasva, 2019-04-15:
hello,I am a member of f.b. homekit italia group and I thank you for the news that constantly introduces in your firmware. I kindly ask you if it is possible to change pause time during the reversing of the shutter motor of the relays in the sonoff dual used like device 12. Currently the pause time is very short, this can cause mechanical problems to the shutter, especially for very heavy ones that have a noticeable inertia. I ask you to bring this time to 500 milliseconds, even better would be until 1 second.Thanks a lot. Max

RavenSystem, 2019-04-16:
@maxnasva Your issue will be better here: https://github.com/RavenSystem/esp-homekit-devices/issues/337

RavenSystem, 2019-05-29:
Finally, this feature will not be implemented. Sorry.

closed 2019-05-29 by RavenSystem