RavenSystem/esp-homekit-devices #331: Firmware 0.8.2 and Inching Function

GPL71, 2019-04-13:
@RavenSystem  Thank you Josè for update
New inching function is fantastic!!!
I have a question. 
I'm using a dual channel board low voltage (geekcreit I believe). 
Firmware work fine. 
I want set 2 different inching time, but I have only one setup. Same inching in both channel?
Channel 1 work the inching
Channel 2 only work only ON/OFF

Do I change something else in setup
Thank you again for marvelous devices you make!!

RavenSystem, 2019-04-13:
@GPL71 Your setup is right. Actually, inching mode only works in channel 1.

I want to update SDK that comes with FreeRTOS 10.2, and then I will declare a secondary timer to manage inching in channel 2. Because I will use 2 different ETSTimers, each channel will can have different times.

GPL71, 2019-04-13:
@RavenSystem Simply Marvelous

i3laze, 2019-04-13:
I’m not fond of the inching implementation, here’s why:
- device acts as _stateless_ switch (like device 13 “lock”), but I would prefer _statefull_ mode: 1st inch = logical On, 2nd = Off
- initial state doesn’t work as intended - it’s currently still Full On, while an inch is expected.

kittmark, 2019-04-14:
Well if it’d be stateful it would be misleading...I prefer it this way, maybe ask for an option to do what you ask, but default should be the way it already is...the way you want it should be reserved for when the pushing the homekit virtual button is absolutely the only way to trigger the relay...and even then, it would get confusing sometimes...

i3laze, 2019-04-14:
I agree about possible confuses and I know @RavenSystem likes backing up states with reliable sensor data (like closed sensor for Garage door e t.c.), but I see no innovation/difference between existing device type 13 and new types added - just HomeKit icons.

I plan using Sonoff SV with external button as the only control, so my scenario has a right to live.

kittmark, 2019-04-14:
Yeah I can see your specific need, but the default should be the way it is (as a person who was waiting for this function for months, I’m certainly happy that it was “beta tested” on the Lock type and then quickly expanded to other types, why should my lights and plugs look like lock icons?). The other way could be even dangerous. But yes, as an option it could be useful in your case.

RavenSystem, 2019-04-14:
Inching can not be statefull because there is not anyy way to know the status of the device and, as @kittmark says, it can be confusing. 

v0.8.3 is released adding inching mode to all channels and at boot.

closed 2019-04-14 by RavenSystem

i3laze, 2019-04-14:
Please, list some of stateless applications (except lock).

I believe relay should control something, that has On and Off state.

i3laze, 2019-04-14:
I’ve just updated several Shelly devices - seems all of them had 0.3sec inch after upgrading from 0.8.1, however I didn’t ever set them as device type 13.

I suggest forcibly cleaning inch to 0 sec on next upgrade, if it equals 0.3 sec.

rechin304, 2019-04-14:
I got same behaveur on Sonoff basic on device type 1

i3laze, 2019-05-02:
I’ve just upgraded v0.5.15 Sonoff Basic via OTA I took from a dusty shelf:
It immediately got  inch 0.3 set.

Please do smth about it.