RavenSystem/esp-homekit-devices #333: Option to disable wifi reset after quick external button pressing?

kittmark, 2019-04-14:
Hello, could you please add in Eve an option to disable the wifi reset that happens when somebody accidentally triggers the external toggle quickly a few times? 
I was testing my just installed Shelly by quickly pressing the external toggle and I accidentally reset the wifi...good thing it’s just the wifi and not factory reset though...I was afraid I had to reopen everything...

I know for sure it can be done because I know another firmware that does it (disable reset sequence for sonoff and shelly).

kittmark, 2019-04-14:
Wait, is the trigger sequence already restricted within the first minutes from powering on?

If so, never mind, it’s already safe enough I guess.

RavenSystem, 2019-04-14:
Yes, reset must be triggered within 2 first minutes.

closed 2019-04-14 by RavenSystem