RavenSystem/esp-homekit-devices #337: DEVICE 12: Improvement suggestion

GPL71, 2019-04-15:
@RavenSystem Josè

Is it possible add following feature?
- delay of 1 or 1.5 second when motor is working and the opposite direction is required (to safe motor and mechanism)

- if motor is moving in a direction and the button of that direction is pressed, then the motor stops

Thank you

seritos, 2019-04-15:
Hi@GPL71.

IMO As long as the code blocks a direction whe the oposite is requested at the same time, i dont think you need taht extra second. This code replicates the behaviour of the mechanical button.

Regarding your second question, this is what it does know...

GPL71, 2019-04-16:
@Seritos thank you!!
About first point 
The delay is only for more safety, when you are working with oldest covering. It is for mechanical and for motor

About second point
I'll check again, but not seem woking in that way

GPL71, 2019-04-16:
@seritos @RavenSystem

ok just tested .. now is working point 2

Then about point 1 only a delay of 1 or 1.5 second (depending of dimension of covering) before inversion

Thank you

RavenSystem, 2019-04-16:
Delay is possible, but difficult. It is necessary to manage what happens while delay, if a button is pushed, or a new order is given.

But I think that delay is not necessary because motors have electro-stop that works immediately when power cuts.  

BTW: @GPL71 I sent you a mail. Did you receive it?

seritos, 2019-04-16:
Thank @Ravensystem for you fast reply. 100% agree with you.

maxnasva, 2019-04-16:
Hello,
Delay is necessary only when you press push botton UP while electric roller shutter is going DOWN,without manually stopping the shutter (and _viceversa).In the shutters control circuits in italy very often an interlocked double button (without a zero position)  is used instead of the switch with zero central position.
In this case (during the reversal) one second -1.000 msec-  as additional time is advisable to avoid high torsions on the shutter shaft.
I hope I was clear.
Thnks 
Max

GPL71, 2019-04-17:
@RavenSystem Josè no email received ...

RavenSystem, 2019-04-17:
@GPL71 I resent you the email. Any news?

GPL71, 2019-04-17:
@RavenSystem Josè nothing again
I have send another one this morning 8.50 to advise.
I'm using the same email which you have wrote me

GPL71, 2019-04-17:
@RavenSystem just sent an email from my icloud account

GPL71, 2019-04-23:
@RavenSystem Josè
The required delay is because in Italy we have external rolling shutter in plastic or alluminium materials.
Then as you can image they have a weight and an effort when it is moving.
To avoid sufference to mechanism and motor, a delay timer (one second) when you reverse it would be good
Thank you

RavenSystem, 2019-06-16:
https://github.com/RavenSystem/esp-homekit-devices/releases/tag/RC_0.9.6

closed 2019-06-16 by RavenSystem