RavenSystem/esp-homekit-devices #341: customize ravencore firmware

franklindee, 2019-04-16:
Hi what if i would like to modify the firmware to my custom needs?

i would like to get a notification everytime the door get open or closed

do you please tell me with wich program i can modify it?
thank you

RavenSystem, 2019-04-16:
You are free to modify firmware, conditions to do are in the License: https://github.com/RavenSystem/esp-homekit-devices/blob/master/LICENSE

But remember that notifications are managed by Apple. You can not add HomeKit notifications to accessories that doesn't have.

franklindee, 2019-04-16:
Hi can you please tell me with wich program i can modify the firmware??

I’ve seen one guy that have modified a sonoff touch connected to a magnet that get notification when the magnet open and close 

At below link
https://youtu.be/KfzeZMIxj-c

I would like to collaborate with you if is possible i’m working on a electrical door lock 
That is connected to this sonoff

Let me know what you think

Thank you

Inviato da iPhone

franklindee, 2019-05-20:
hola Jose
que tal? i'm really following your job and i really appreciate the fact that you are opening the homekit world to everybody...i would really like to follow your step helpin to delvelop new thing.. but i rellay would like to start since the beginning and if you would give me some tips i would appreciate...

amazing job what you are doing with the HAA,

attached some of the projects i would like to realize with your firmware

i would like to create a door accessory with electronic lock for my fence and my main door 

can you please support me on that i would dedicate my time to study all the thing and be able to program a dedicated firmware?
thank you

Francesco Dragone

RavenSystem, 2019-05-20:
@franklindee Hello,

First of all is to download official HAP documentation from Apple: 
https://developer.apple.com/support/homekit-accessory-protocol/

Then, you can see simple examples here:
https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples

You can begin with LED example:
https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/led

And here you have a lock example:
https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/lock

To know how characteristics and services work, yon can read Apple HAP documentation and this header file:
https://github.com/maximkulkin/esp-homekit/blob/master/include/homekit/characteristics.h

closed 2019-05-26 by RavenSystem

franklindee, 2021-05-05:
Hi Jose,
i was working on a sonoff 4ch pro r3 to use as programmer for the irrigation system and i've used a tested json in your issues on git up i just noticed that there were some errors so i've corrected it

{"c":{"o":0,"m":60,"l":13,"i":0,"f":10,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":20,"b":[{"g":0,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":20,"b":[{"g":9,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":20,"b":[{"g":10,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":20,"b":[{"g":14,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]}]}

let me know if you need other info

thank you

Distinti Saluti
Francesco Dragone
+393400607091
***@***.***