RavenSystem/esp-homekit-devices #345: Device 14: Magic Home enhacement

Partizan7676, 2019-04-18:
RGB controllers can be used for control 3 different led strips(no colour) - you can connect them to each color. Would be fine if device 14 can convert to 3 led lamps, controlled with homekit. On/off and brightness.
Also is great to add support for WS8212 LEDs. in https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/led_strip_animation it works fine and can be used as option in device 14, or new device type.

GPL71, 2019-04-18:
it seem working on electrodragon too, but some adjustment is necessary.
And it seem suitable for follow bulb, Im reading to understand better

http://tinkerman.cat/ailight-hackable-rgbw-light-bulb/#lightbox-gallery-AGJ2n8kZ/12/

jakovbutko, 2019-04-24:
Hey. Can I add support for IR remotes in your firmware? 
What I could find:
  { 
  "MagicHome RF",    // Magic Home RF (ESP8266) - (Arilux LC10)
     GPIO_USER,        // GPIO00 Optional Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED1_INV,    // GPIO02 Blue onboard LED (optional)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor0
     GPIO_ARIRFRCV,    // GPIO04 RF receiver input 
     GPIO_PWM2,        // GPIO05 RGB LED Green
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM3,        // GPIO12 RGB LED Blue
     GPIO_PWM4,        // GPIO13 RGBW LED White
     GPIO_PWM1,        // GPIO14 RGB LED Red
     GPIO_LED2_INV,    // GPIO15 RF receiver control
     0, 0
  },

RavenSystem, 2019-05-03:
> Hey. Can I add support for IR remotes in your firmware?

Sorry, IR remotes never will be supported.

closed 2019-05-10 by RavenSystem