RavenSystem/esp-homekit-devices #347: New simple feature for device nr. 10

lizzus, 2019-04-18:
Hi Josè, I wanted to ask you if in the next release you can insert a small modification to the device nr. 10 (ESP01 Switch + Button). I have heard so many people who would like only one button, without the relay, to create wireless buttons, even considering the reduced dimensions of esp01. 
Currently the device 10 has both button and relay and I thought it would be very convenient to give people the possibility of having only 1 device, simply setting it in EVE. 
In this way with the 
setting 0 = socket + button (default)
o
setting 1 = button only
People can choose the solution that best suits their needs. Of course after the restart the devices will also be updated at home.

GPL71, 2019-04-18:
I agree it would be interesting

curla92, 2019-04-19:
> I agree it would be interesting

me too

GuyCoast, 2019-04-19:
idem

vglucas, 2019-04-21:
very interesting, it interests me too 👍

lizzus, 2019-04-22:
hi @RavenSystem didi you read?

maxnasva, 2019-04-22:
oh yes,it's very interesting.I hope this change is not too difficult to include in the firmware.

GPL71, 2019-04-23:
It would be interesting if you need add an automation for a group of devices.
An example about:

1 stateless button on wall-switch
2 different devices with their relay and bulbs connected

Then
Press 1 time button and device 1 turn on
Double press and device 1 turn off, while device 2 turn on
Long press and devices 1 and 2 turn on

It imitate the relays mad by FINDER

What do you think

maxnasva, 2019-04-23:
commutador ref.27.05/25 finder
[image]

maxnasva, 2019-04-23:
commutador finder ref.27.05/25
https://gfinder.findernet.com/public/attachments/27/ES/S27ES.pdf
[image]

RavenSystem, 2019-04-30:
@lizzus With HAA you will can create an event button. I don't want to improve dev type 10 anymore. Sorry.

lizzus, 2019-04-30:
event o stateless button?

RavenSystem, 2019-04-30:
stateless button, same button with dev type 10 (without switch and relay).

RavenSystem, 2019-05-01:
Try version 0.8.7 with dev type 15 and configure pin w GPIO with your button.

GPL71, 2019-05-01:
It working!!!
Thank you so much

lizzus, 2019-05-01:
works fine. thanks!

closed 2019-05-01 by RavenSystem