RavenSystem/esp-homekit-devices #348: REQUEST - DTH22 wake

claire078, 2019-04-19:
Hi,
I use this sensor with external power supply (because I have only two wires in wall), works good but when I have a electric power fall then it shows 0.0C, and I have to sensor power off then on. 
Is there any software solution for this?

claire078, 2019-04-19:
It doesn’t work good neither. 😕
Sometimes temperature drops to ~10C. I tried 4,7K resistor between + and data legs. Doesn’t help.
I use sensor with long ~6m cable.

pixidixi88, 2019-04-20:
Hi, I use also long cables and I have the same issue. Not always is 0.0 sometimes is last reader temperature but it is green and not heat even target is higher. To resume functionality I need to disconnect sensor and connect it again.

claire078, 2019-04-22:
The main problem is when shows 9.9C then heats because it’s lower then set point. 😕
I have to figure out another solution. 🤔

RavenSystem, 2019-05-03:
I need to test it and try to replicate this issue.

mafyata, 2019-05-16:
Did you use nodemcu + dht22

claire078, 2019-06-03:
Nodemcu? I use Sonoff Basic. 
I have a similar problem with less than 1m cable. It shows grayed temperature value in Eve app and looks it doesn’t update it.

claire078, 2019-06-04:
I tested in device 6. and it works good. But I need a thermostat.

claire078, 2019-06-04:
No, it doesn’t work with device 6. After a while it shows last value, after setup on it shows 0 celsius and 0% humidity. After power cut it works again. 
Maybe sensor fault?

mafyata, 2019-06-04:
Mine is working fine. Device type 6, board type 1, th sensor type 2.

claire078, 2019-06-04:
Mine board type 2, sensor type 2 DHT22. 
What could be than the problem? 😕

mafyata, 2019-06-04:
Try board type 1 if you are not 100% sure that it is board type 2. May be the sensor is bricked.

claire078, 2019-06-04:
I’m 1000% sure that is board type 2. Again sensor works, but after a while it stops working.

mafyata, 2019-06-04:
See update period in eve app

mafyata, 2019-06-04:
Mine is 30s

claire078, 2019-06-04:
Mine 10sec..

closed 2019-07-20 by RavenSystem