RavenSystem/esp-homekit-devices #350: 0.8.5 Dummy switch

kalety, 2019-04-21:
Hi Jose!, I've updated one of my switches to 0.8.5 version. At Eve's app appears the dummy switch option, but no idea how to manage it, not appears at Home app. Can you explain in more detail how works it?.

Thanks in advance!

RavenSystem, 2019-04-21:
I am busy now, but give me a chance to write documentation about it in Wiki.

lizzus, 2019-04-21:
Please after link me where you add, so I can add in my Italian translation. Thanks.

RavenSystem, 2019-04-21:
Instructions about Dummy Switch:

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#dummy-switch-default-off

RavenSystem, 2019-04-21:
@lizzus Check https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#pins-r-g-b-w too

lizzus, 2019-04-21:
Ok, thanks.

RavenSystem, 2019-04-21:
@lizzus If you want, I can add a link to your Italian translation in the Wiki.

lizzus, 2019-04-21:
Really? Yes, off course!

RavenSystem, 2019-04-21:
@lizzus What is the url?

lizzus, 2019-04-21:
https://www.twibuk.com/read-blog/22_guida-completa-in-italiano-del-firmware-ravencore.html

RavenSystem, 2019-04-21:
@lizzus Done!!

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore

closed 2019-04-21 by RavenSystem

kalety, 2019-04-21:
> Instructions about Dummy Switch:
> 
> https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#dummy-switch-default-off

Mil gracias!

i3laze, 2019-04-24:
Great dummy!
I’d personally use it with Garage door (instead of extra dummy from HomeBridge) to overcome the need of unlocking iPhone before “someone arrives” automation or saying “Open garage door”.

However I still willing to see Inching relay saving it’s logical state (1st click on, 2nd off) in such a dummy, actually.