RavenSystem/esp-homekit-devices #351: Garage door open/close sensors not working

fluppie, 2019-04-22:
Hi,

The firmware (0.8.4) itself is working fine. The relay toggles (in my case a PC817 optocoupler) and the gate opens or closes.
I also have connected open and close sensors (LJ12A3-4-Z/BX trough a PC817 optocoupler). I connect the GPIO's to GND to give a logical one. With ESP Easy and Tasmota I can read out the status of the sensors.
GPIO 12 is open sensor
GPIO 13 is closed sensor
GPIO 14 is relay/optocoupler toggle

I did change the following to main.c
```
// Sonoff
#define LED_GPIO                        0

#define DOOR_OPENED_GPIO                12
#define DOOR_OBSTRUCTION_GPIO           5
#define DOOR_CLOSED_GPIO              	13

#define COVERING_POLL_PERIOD_MS         250

#define TOGGLE_GPIO                     3

#define BUTTON1_GPIO                    0
#define BUTTON2_GPIO                    9
#define BUTTON3_GPIO                    10
//#define BUTTON4_GPIO                  3

#define RELAY1_GPIO                     14
#define RELAY2_GPIO                     5
#define RELAY3_GPIO                     4
#define RELAY4_GPIO                     15
```
I used the eve app to configure everything, but cannot get it to work.

To be sure, is RavenCore programmed to tie the GPIO to 3.3V or GPIO to GND? Maybe the issue lies in that part since I can detect open/close with Tasmota and ESP Easy?

fluppie, 2019-04-22:
For those who care about the HW side https://wiki.wemos.cc/_media/products:d1:d1_v2.0.0.pdf
[image]

i3laze, 2019-04-24:
I personally use Sonoff with NC reed sensor attached between GPIO14 and Ground, and sensor does work  0.8.4.
However pay attention to “Sensor close NC” parameter, as it determines how the sensor is actually used. Set wrong you will see meaningless states.

fluppie, 2019-04-25:
I cut the 3V3 pull up resistor (on the left of my opto-couplers) and now the OPEN sensor is changing the status. Only the CLOSE sensor is not (yet?) working as expected. My garage door does open-stop-close when you press the single button on the remote. I don't think the software is understanding these kind of systems. Could we use a timer to determine how far open/closed the garage door is with the working time?

fluppie, 2019-04-25:
OK, got it. Changed lines 2032 till 2040
before: extra_gpio ==> now: DOOR_CLOSED_GPIO

```
            adv_toggle_create(extra_gpio, true);
            
            if (custom_garagedoor_sensor_close_nc.value.bool_value) {
                adv_toggle_register_callback_fn(extra_gpio, door_closed_1_fn_callback, 0);
                adv_toggle_register_callback_fn(extra_gpio, door_closed_0_fn_callback, 1);
            } else {
                adv_toggle_register_callback_fn(extra_gpio, door_closed_0_fn_callback, 0);
                adv_toggle_register_callback_fn(extra_gpio, door_closed_1_fn_callback, 1);
}
```

fluppie, 2019-04-25:
OK, it working well, only when (re)booting the module the relay clickers and so the garage door is opening... After boot the relay stays in state 1 instead of 0. Once you did run one command for example, open the door it's all good. But this reboots can happen when, WiFi outage, power outage etc.

RavenSystem, 2019-05-03:
You don't need any extra hardware to manage sensors. The software has all you need.

You don't need to change code. I think that you are using a wrong board type.

Sensors must be connected to GPIOs and GND.

MarcioPlaza, 2019-05-04:
Hi, fine, I have a basic sonofound for use in my garage. It turns out I am wanting to use GPIO RX or TX for open door sensor, is it possible to do this using eveapp? if not how can I solve this. I tried to solder GPIO4 but it is very small. If I have no other solution how do I reconfigure the code generating a .bin file? What program do you recommend? I'm a beginner in programming, sorry for the many questions.
Thank you...

MarcioPlaza, 2019-05-04:
Hi, fine, I have a basic sonof basic for use in my garage. It turns out I am wanting to use GPIO RX or TX for open door sensor, is it possible to do this using eveapp? if not how can I solve this. I tried to solder GPIO4 but it is very small. If I have no other solution how do I reconfigure the code generating a .bin file? What program do you recommend? I'm a beginner in programming, sorry for the many questions.
Thank you...

i3laze, 2019-05-04:
Basically you need just Close sensor on GPIO14.
Fully opened sensor is optional.

MarcioPlaza, 2019-05-04:
I'm already using GPIO14 but when I open the gate the open status does not work even though I set the warning time.

i3laze, 2019-05-04:
Did you try changing Normally Closed (NC) option for your sensor?

mientki, 2019-05-05:
It seems to me that there is a logical error in RavenCore. The "OPEN SENSOR NC" setting is missing and the accessory can not be set correctly. In the case of the Garage Door configuration and the use of two contact sensors, the accessory can not be configured to behave as follows:

We have 2 fixed contact sensors and one magnet mounted on a garage door that moves between these sensors:
1. open sensor - when the sensor is closed, the gate is open
2. closing sensor - when the sensor is closed, the gate is closed

[image]
[image]

1. THE GATE IS CLOSED
- GPIO14 sensor is closed
- The GPIO4 sensor is open

2. THE GATE IS OPENED
- The GPIO14 sensor is open
- The GPIO4 sensor is open

3. THE GATE IS OPEN
- The GPIO14 sensor is open
- The GPIO4 sensor is closed

4. THE GATE IS CLOSED
- The GPIO14 sensor is open
- The GPIO4 sensor is open

return to point 1. THE GATE IS CLOSED

GPIO4 should be set in the opposite direction - when the sensor is closed, the gate status is OPEN. Currently GPIO4 - when the sensor is closed, the gate status is CLOSED.

RavenSystem, 2019-05-05:
> The "OPEN SENSOR NC" setting is missing

@mientki please, read the wiki before posting:

[image]

mientki, 2019-05-05:
I omitted this setting. I set it to 2. Now everything works as it should. Thanks :)

RavenSystem, 2019-05-05:
@mientki ;)

closed 2019-05-05 by RavenSystem