RavenSystem/esp-homekit-devices #353: Shelly 1 as a Water valve

tohmc, 2019-04-24:
Hi,

when I set up my shelly 1 to Device type 7
it have no function.
I think its on other GPIO as in Shelly.
it is possible to add a GPIO setting or when setting to Shelly board type that it uses Shelly GPIO?

Aaronwithanat, 2019-04-28:
The shelly 1 GPIO cannot and should not be used while connected to mains voltage. The GPIO headers on the shelly are directly tied to the AC lines.  Specifically the ground on the header is tied directly to the hot wire or line voltage pin from the AC input.

tohmc, 2019-04-29:
Thanks But  i know this.

Shelly is Jumped to 12V 
But no relay is triggered when connecting it and test it.

tohmc, 2019-04-29:
Nobody an idea?

RavenSystem, 2019-05-03:
Shelly1 is not supported as Water Valve yet.

tohmc, 2019-05-04:
Thanks,

Will ist ne supported in the Future?

RavenSystem, 2019-05-04:
Yes, sure, and soon, because I will use it in my garden.

RavenSystem, 2019-05-11:
0.8.8 supports it.

closed 2019-05-11 by RavenSystem