RavenSystem/esp-homekit-devices #354: Garage door use with no sensors?

jat80, 2019-04-26:
Hi there,

I've just successfully flashed my Sonoff SV with this firmware and configured it for a garage door.  I'm still waiting for my sensors to arrive.  I'm just wondering if I can use it currently without any sensors?

Nothings connected to the garage door yet but when I operate it currently the garage door icon says opening in home app but won't change state until I press the button again then it switches straight back to closed or closing without hearing the relay click on the Sonoff sv.  Is this normal without having sensors connected?

Thanks in advance.

JT.

pilot1981, 2019-04-26:
close sensor is mandatory for this firmware...

peros550, 2019-04-26:
In similar implementations (Homebridge plug-ins) I’ve seen they emulate the opening / closing state using an internal timer. For example , if you give the open command, the accessory will change state to “opening” for a number of predefined seconds and then change again into “open” state. That’s a good workaround when you don’t have reed sensors. You only need to know how much time your Door takes to open/close.

RavenSystem, 2019-04-30:
As @pilot1981 says, close sensor is mandatory.

closed 2019-04-30 by RavenSystem