RavenSystem/esp-homekit-devices #355: Magic Home issue? -maybe not-

greensouth, 2019-04-27:
Hi Jose!

It's possible that since I applied Ravencore to a RGB Magic Home it's bright is lower than the original firm?

Maybe it's not, but I would swear that it's about 10% less bright.

I fiddled with the color boost option but it seems it only lower even more.

Maybe there are demons in my controller!

Also, a couple of times I lost connection and had to reflash it in order to link it to my wifi again. Tried to perform a reset (10 reboots in less than 3 seconds.... But it is impossible when you have to switch it on and off pulling the cable xD)

Thanks! And great exposition at Reddit

RavenSystem, 2019-04-27:
> Hi Jose!
> 
> It's possible that since I applied Ravencore to a RGB Magic Home its bright is lower than the exposed with the original firm?
> 
> Maybe it's not, but I would swear that it's about 10% less bright.
> 
> I fiddled with the color boost option but it seems that it only lowered even more to me.

Set color boost to 100% and select a different color. Color boost value is applied in the next color change. Tell me your results, please.
 
> Maybe there are demons in my controller!

Demons are always welcome ;)

> Also, a couple of times I lost connection and had to reflash it in order to link it to my wifi again. Tried to perform a reset (10 reboots in less than 3 seconds.... But it is impossible when you have to switch it on and off pulling the cable xD)

If you lose wifi connection, you don't need to refresh. Try to put your device is other place with better wifi coverage and repower it. Actually, you can not reset RGBW device switching it on and off. The only way is to push and hold the button (GPIO0). In next version, you will be able to reset any accessory switching it quickly from Home App too.

> Thanks! And great exposition at Reddit

Cheers! :)

mafyata, 2019-04-27:
> Hi Jose!
> 
> It's possible that since I applied Ravencore to a RGB Magic Home its bright is lower than the exposed with the original firm?
> 
> Maybe it's not, but I would swear that it's about 10% less bright.
> 
> I fiddled with the color boost option but it seems that it only lowered even more to me.
> 
> Maybe there are demons in my controller!
> 
> Also, a couple of times I lost connection and had to reflash it in order to link it to my wifi again. Tried to perform a reset (10 reboots in less than 3 seconds.... But it is impossible when you have to switch it on and off pulling the cable xD)
> 
> Thanks! And great exposition at Reddit

Do you have original software of magic home rgb controller ? I forget to backup mine before flash it.

greensouth, 2019-04-27:
Nope, same bright but color changing. I suppose it's happening because I have no white channel. But anyways still low bright.

When set boost at 100%, some colors just blinks and turns off.

Maybe voltage passed to the strip is not enough? I don't know if this is managed when programing a firmware for the controller.

:(

RavenSystem, 2019-04-30:
> Nope, same bright but color changing. I suppose it's happening because I have no white channel. But anyways still low bright.
>
> When set boost at 100%, some colors just blinks and turns off.
>
> Maybe [...]

Don't worry about white channel.

Try this: With brightness at max and color boost 100%, set pure red color, and same with pure blue and pure green; and post here the results, please.

RavenSystem, 2019-05-07:
@greensouth Any news about that? Did you try what I said?

greensouth, 2019-05-07:
Wenas José!
Actually I have nothing installed on my iPhone which let me set RGB values manually or pure colors, I move the color picker as close as I think it's the most pure values in both EVE and Home App. Still see them a bit dim. I think I was able to set color boost to about 30% to get the maximum fidelity of colors/brightness with my RGB controller. 

I suppose that it's having some difficult matching the color wheel (which uses WHITE as a color to mix all the possible combinations) with only 3 channels. 

I'm expecting to receive some RGBW Led stripes + Controller this week and I'm planning on flashing one with Ravencore and let the other with the default firmware and I will check both.

I'm curious about what is exactly the function of color boosting when getting / setting the color values.

Btw, were you able to take a look at what I told you about adding some custom delay for the timers when beginning the movement of the rolling shutters? 

Thanks!

RavenSystem, 2019-05-10:
@greensouth 

If you want a good white color, RGBW led strip is your best choice. Mixing RGB to get white is... well... is a shit.

Color boost is a factor from 1 to 3 (0% to 100%) to mix RGB colors and try to match with your led strip in a better way.

> Btw, were you able to take a look at what I told you about adding some custom delay for the timers when beginning the movement of the rolling shutters?

I remember something about this, but I am not able to find it now. Can you post the link of the issue?

greensouth, 2019-05-10:
Hey @RavenSystem I've just received the rgbw strip and WORKS LIKE A CHARM! With Ravencore. I'm changing all my RGB stuff.

The roller shutter thing is a little miss-adjustment when setting the time that takes to open and close the rolling shutter.

It takes a couple of seconds, maybe three, to tight the hooks before it begins to move. That adds some extra time where it's not moving at all but enough to break the calculations for setting %. In my case, closing at 50% is far from the half of the window.

Adding extra time for opening/closing doesn't work. It would be needed to add a setting for adding a configurable delay for 0 and 100 positions, before starting to count the real time once the hooks are tight, which is when the rolling shutter begins to move.

Maybe it's a bit difficult to explain :/

RavenSystem, 2019-05-18:
@greensouth Más cómodo en español:

Sé que te refieres a las persianas que tenemos todos en nuestras casas, al menos en España, que las puedes bajar pero dejando las ranuras esas abiertas para que entre algo de luz, o aire. Obviamente, el % de HomeKit se refiere realmente a la posición del motor, y no de la persiana en sí.
Al 0%, la persiana está totalmente cerrada, sin ranuras abiertas ni nada, en modo búnker.
Al 10%, la persiana seguro que seguirá cerrada, pero con las ranuras abiertas.
Y al 50%, la persiana estará un poco abierta, pero no a la mitad. Aunque el motor sí que estará a su mitad de recorrido.

Poniendo tiempos más altos de lo que realmente tarda la persiana en subir y en bajar no sirve para nada. De hecho, es mejor poner el tiempo exacto, ya que luego esto sirve para que en los casos en los que pierda un poco de sincronización (estos motores no tienen unos tiempo de trabajo exactos), el accesorio pueda volver a sincronizarse automáticamente cuando subas o bajes la persiana del todo.

Ahora bien, tú sugieres poder indicar el tiempo que tarda la persiana de estar totalmente cerrada, a estar cerrada pero con las ranuras abiertas, y que empieza a contar el porcentaje una vez que ha alcanzado esa posición. Así si le dices que la abra al 50%, se abrirá al 50%, pero ¿qué pasa si quieres poner la persiana con las ranuras medio abiertas, o solo un poco por la parte de arriba? Yo a veces las quiero así. No te creas que no le he dado ya vueltas a este tema, y sigo en ello. Por ahora la mejor idea que he tenido es que del 0 al 5% corresponda a cuando la persiana está de totalmente cerrada a cerrada pero con todas las ranuras abiertas. Y ya del 5% al 100%, sería de la persiana cerrada (ranuras abiertas), a la persiana abierta. De esta manera, el 50% estaría muy cerca de la mitad real.

greensouth, 2019-05-18:
Es que parece una chuminada, pero al final tiene la cosa más enrea de lo que parece. Todo tiende a liarse ;) además veo que estás a tope con el otro proyecto.

Te va a echar el coco lumbres!! 

X cierto el tema Magic Home he estado dándole unas cuantas vueltas más, y cuando hace el Fade al cambiar de color, sube a tope la intensidad y luego cuando llega al color final, es cuando la baja. 

Ahí es donde está la pequeña pérdida de brillo que te comentaba. 

;)

RavenSystem, 2019-05-18:
Por el otro proyecto no te preocupes. En realidad tengo un interés personal en el tema de las persianas. ¿Qué te parece lo de dejar un 5% para la parte de las ranuras?

Respecto a lo de los LEDs, es posible que pase lo que dices. Mejoraré el algoritmo, pero puedo asegurarte que el color final  que te pone ya fijo es el color seleccionado.

greensouth, 2019-05-18:
Acabo de probarlo y a mí se me empiezan a ver todas las rajitas a partir de casi el 10%, pero todo esto es muy elástico, supongo que dependerá del motor, de los flejes y tal.

RavenSystem, 2019-05-18:
Claro, por eso digo de configurar el primer 5% aparte con el tiempo que tarda el motor en poner la persiana a punto para empezar a abrirse (ya con todas las ranuras abiertas).

Ya que estás, haz una cosa: Pon la persiana en el % que sea para dejarla en el punto justo antes de empezar a abrirse, y dime en qué % es.

RavenSystem, 2019-05-18:
Y prueba la version 0.8.11 que acabo de sacar con lo de los LEDs y me cuentas.

greensouth, 2019-05-18:
En mi motor sobre el 10%

RavenSystem, 2019-05-18:
> En mi motor sobre el 10%

Entonces si le dices que te ponga la persiana al 55% tendría que estar abierta a la mitad, ¿no?

greensouth, 2019-05-18:
Tal cual está ahora mismo, para notar que está subida al 50% , tendría que ponerla al 70% sin exagerte

RavenSystem, 2019-05-18:
> Tal cual está ahora mismo, para notar que está subida al 50% , tendría que ponerla al 70% sin exagerte

¿Y tienes los tiempos exactos puestos? (De un final de carrera a otro del motor)

greensouth, 2019-05-18:
Si tío, es súper raro, pero tb lo monté en casa de un vecino con el mismo motor y tal, y le pasa igual, de 0 a 100 cuadran los tiempos con los finales de carrera, pero visualmente por lo que te comenté hasta que se tensan los flejes, se queda muy corto en los porcentajes intermedios.

RavenSystem, 2019-05-18:
Ahhh, claro. Ya sé por qué pasa eso. Cuanta más persiana hay recogida, más radio tiene el eje, y más longitud de persiana recoge por vuelta.

greensouth, 2019-05-18:
Le vas a tener.que meter un multiplicador o algo! Jajajajaja

RavenSystem, 2019-05-18:
Ya pensaré un algoritmo para solucionar esa chapuza de controlador de persianas, y digo chapuza por ser suave, porque es una m... bueno, esta es la parte divertida de la programación.

Y no te olvides de probar la 0.8.11 con los LEDs.

greensouth, 2019-05-18:
Mañana lo pruebo con el que más problemas me da, porque cuando lo flashee no me dejó hacerlo con OTA (no sé si ese día estaría el servidor caido o algo)  y tengo que volver a soldar e intentarlo a ver si esta vez me deja la actualización :/ 

Oye muchas gracias por el currazo que te estás pegando. Estoy deseando de que me llegue el NodeMCU a ver si tengo huevos de echar a andar y probar lo otro!

RavenSystem, 2019-05-18:
De nada hombre. Yo me alegro de que mis firmwares sirvan para algo medio útil, aunque como controlador de persianas sea chapuzero (aún). Y yo odio las chapuzas.

RavenSystem, 2019-05-19:
@greensouth Actualiza alguna de tus persianas a la 0.8.12, ciérrala del todo, y luego entra en EVE y pon la opción "Boost" al 100% (Es una opción nueva), y prueba a ver qué tal van ahora las posiciones intermedias.

https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore#boost-default-0

greensouth, 2019-05-19:
Justamente lo estaba haciendo!!

Con el 71% de boost LO-CLA-VA

Ahora es PERFECTO!!! Muchas gracias tiaco!!!!!! Fíjate si estoy contento que en cuanto cobre (y pague las rift s) tiro a pillar el motor para el salón !!

Ayer estuve dándole a los LED y he llegado a la conclusión de que la transición entre colores satura mucho al controlador, y le mete lagazos, y más cuando usas la rueda de HomeKit, que es impreciso y te hace moverte muchas veces entre colores en un espacio muy corto de tiempo,  creo que la transición animada debería ser opcional con un switch, qué opinas?

mafyata, 2019-05-20:
Can someone tell me which command to use to download original magichome firmware before flash with ravencore ?

RavenSystem, 2019-05-20:
@mafyata `esptool.py -p /dev/espport read_flash 0x0 firmware.bin`

mafyata, 2019-05-20:
Where will be downloaded ? Is only one file to download?

RavenSystem, 2019-05-20:
> Where will be downloaded ?

In the working directory.

> Is only one file to download?

Yes, only a file with a full firmware backup. Then you only must write it at 0x0 address.

mafyata, 2019-05-20:
I don’t know the directory, may be esptool directory. Thank you very much. :)

RavenSystem, 2019-05-20:
> I don’t know the directory, may be esptool directory. Thank you very much. :)

The working directory is where you run the command. Instead, you can run:
`esptool.py -p /dev/espport read_flash 0x0 C:\firmware.bin`

mafyata, 2019-05-20:
This command will do the job easier. Thanks again

RavenSystem, 2019-05-20:
@mafyata With that command, you can do a backup of any firmware, not only of Magic Home.

mafyata, 2019-05-20:
🥇

mafyata, 2019-05-21:
Problem with reading flash:
C:\WINDOWS\system32>esptool.py -p com4 read_flash 0x0 firmware.bin
usage: esptool read_flash [-h] [--spi-connection SPI_CONNECTION]
                          [--no-progress]
                          address size filename
esptool read_flash: error: argument size: invalid arg_auto_int value: 'firmware.bin'

mafyata, 2019-05-21:
I find an issue:
esptool.py -p com4 -b 115200 read_flash 0 0x200000 c:/firmware.bin

RavenSystem, 2019-05-27:
@greensouth Prueba la versión 0.8.13 a ver si ya no te da tirones la transición de colores.

greensouth, 2019-05-27:
Te cuento,

En el cambio de color es como si partiera no del color inicial, sino de un blanco o algo más intenso.

Si estoy en rojo y muevo al azul, hace algo así como ROJO - BLANCO - AZUL. Estoy intentando subirte un vídeo a ver si lo puedes ver, y gratis y por cortesía, las conversaciones de mis vecinos que se escuchan por el lavadero. 🤪

https://www.youtube.com/watch?v=KPsYG5qgAWY&feature=youtu.be

RavenSystem, 2019-05-27:
@greensouth Ahora lo veo claro, reproduciendo el vídeo al 25% de su velocidad. En problema es que funciona demasiado bien, es decir, la respuesta es tan inmediata al cambio de color, que junto con la porquería de rueda de selección de color de la aplicación Casa, da como resultado una verdadera m... Fíjate que en cuanto tu pones el dedo sobre la rueda ya está aplicando el color seleccionado, que es más blanco que el que tú realmente quieres, y cuando tú pones el dedo, te aparece la lupa de ajuste fino, y tú desplazas el dedo hacia el exterior de la rueda, pero antes de empezar a desplazarlo, ya te ha cambiado el color al seleccionado nada más posar el dedo. No sé si me explico. El caso es que ya sé cómo solucionarlo.

greensouth, 2019-05-27:
> @greensouth Ahora lo veo claro, reproduciendo el vídeo al 25% de su velocidad. En problema es que funciona demasiado bien, es decir, la respuesta es tan inmediata al cambio de color, que junto con [...]

si es que al final una imagen... ;) Justo ahora estoy flasheando otro magic home que me ha llegado

RavenSystem, 2019-05-27:
Pues vas a tener que actualizarlo en breve, porque en unos minutos saco nueva versión.

RavenSystem, 2019-05-27:
@greensouth Prueba la 0.8.14 a ver qué tal.

greensouth, 2019-05-27:
> @greensouth Prueba la 0.8.14 a ver qué tal.

en ello está, me ha pillado la actualización en mitad del flasheo y no lo veía Homekit, tenía que haber echado la Bonoloto!

RavenSystem, 2019-05-27:
@greensouth Cuando dices "muchísimo más estable" quieres decir que aún no es totalmente estable? O ya sí lo es?

Y sí, puede que me haya columpiado con el segundo de retardo, pero eso tiene fácil arreglo.

greensouth, 2019-05-27:
> @greensouth Cuando dices "muchísimo más estable" quieres decir que aún no es totalmente estable? O ya sí lo es?
> 
> Y sí, puede que me haya columpiado con el segundo de retardo, pero eso tiene fácil arreglo.

eso es que va fino filipino, ya no parece que estés empastillao en una discoteca! ;)

RavenSystem, 2019-05-27:
> eso es que va fino filipino, ya no parece que estés empastillao en una discoteca! ;)

@greensouth Mmm, y si le pongo una opción adicional que sea modo disco y se ponga a poner colores aleatorios? De hecho se podrían precargar varios modos: color fijo, brisa mañanera, lumbre de hogar, bar de luces de colores...

greensouth, 2019-05-27:
> @greensouth Mmm, y si le pongo una opción adicional que sea modo disco y se ponga a poner colores aleatorios? De hecho se podrían precargar varios modos: color fijo, brisa mañanera, lumbre de hogar, [...]

... lupanar, casa de pelos... hay tantas posibilidades! jajajajj de hecho la app del magic home tiene alguna movida de esas.

RavenSystem, 2019-05-27:
@greensouth pues puede que cuando tenga tiempo y ganas lo haga. Por cierto, nueva versión 0.8.15 con el retardo a 300ms, a ver qué tal.

greensouth, 2019-05-27:
> @greensouth pues puede que cuando tenga tiempo y ganas lo haga. Por cierto, nueva versión 0.8.15 con el retardo a 300ms, a ver qué tal.

actualizando!

greensouth, 2019-05-27:
Vaya ahora no conectan a homekit... se le ha ido la olla a todos.

RavenSystem, 2019-05-27:
Vaya cagada, que no había llegado a subir el firmware... ya está arreglado.

RavenSystem, 2019-05-27:
@greensouth Has podido actualizar ya?

greensouth, 2019-05-27:
Acaba de terminar hasta el que me daba problemas para flashear, era todo eso, la ruta del main, y va por fin todo genial, sin flashes, sin locuras. Muchas gracias!!!

Oye tengo curiosidad, si se atasca al actualizar, se queda buscando hasta que esta disponible el fichero? O como funciona en caso de quedarse a mitad de una descarga OTA?

RavenSystem, 2019-05-29:
Cierro incidencia, puesto que ya está solucionado el problema inicial de los LEDs.

> Oye tengo curiosidad, si se atasca al actualizar, se queda buscando hasta que esta disponible el fichero? O como funciona en caso de quedarse a mitad de una descarga OTA?

@greensouth Si se atasca actualizando se pueden dar 2 situaciones:
1.- Que no se haya empezado a descargar la actualización: En este caso, basta con apagar el cacharro, volverlo a encender  y decirle de nuevo que actualice.
2.- Que ya haya empezado a descargar la actualización: apagar el cacharro, volverlo a encender y dejar que continúe la actualización por sí mismo.

Se puede ver el progreso de la actualización/instalación OTA usando el siguiente [comando de NetCat](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#lcm-ota-logs) en macOS o GNU/Linux:
`nc -kulnw0 45678`

Si quieres saber más acerca de cómo funciona el sistema LCM (OTA), puedes preguntar tus dudas en su repositorio https://github.com/HomeACcessoryKid/life-cycle-manager . @HomeACcessoryKid te responderá gustosamente a todo.

greensouth, 2019-05-29:
Pues esto te va a encantar, por cerrar el hilo! Con la intensidad ocurre algo parecido. 
[https://www.youtube.com/watch?v=hihsIyb47wo&feature=youtu.be](url)

mafyata, 2019-05-29:
Can’t open video link

RavenSystem, 2019-05-29:
@greensouth Pues eso sí que no es normal.

greensouth, 2019-05-29:
No te olvides de subir el main.bin! ;)

RavenSystem, 2019-05-29:
@greensouth Jajajaja, ya lo he revisado, por si acaso. Por cierto, lo del parpadeo ese no me lo explico. EL blanco es manejado igual que los otros colores. Puede que sea por falta de potencia. ¿Puedes hacer más pruebas a ver si te pasa con todas tus tiras LED, o con otros colores?

[27 more replies]

closed 2019-07-02 by RavenSystem