RavenSystem/esp-homekit-devices #356: Home and Eve apps doesn't see the switch

kkwwm, 2019-04-27:
I have a problem adding Sonoff Touch T1 to the Home and Eve app. The switch is flashing correctly, it is visible in Eve, but it does not pair. The Home app does not see the switch at all. I was flashing Sonoff Basic in the same way and everything went well. What can I do to use my Touch T1 with RavenCore and homekit???
Thank you for your help!

bbfelts, 2019-04-28:
I'm having a similar issue with Shelly 1's.  They flash successfully, get through the pairing process in the Eve app, but then become non-responsive regardless of what I do. I can seem them in both the Eve app and Home app but they are non-responsive.  I looked at the logging and there aren't any errors.  Restarting the devices (turning off power and turning back on) doesn't seem to help.

peros550, 2019-04-28:
In case you haven’t tried so, try erasing flash before flashing.

bbfelts, 2019-04-28:
I have tried completely starting from scratch (erasing, refreshing, setting up wifi) and the same happens.  I was able to capture some logging of the pairing process and I see some socket errors (IDs changed for protection):

>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 6
>>> HomeKit: Got new client connection: 7
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 8
>>> HomeKit: [Client 7] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with XXXXXXXX-YYYY-ZZZZ-AAAA-BBBBBBBBBBBB
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 8] Pair Verify Step 1/2
>>> HomeKit: [Client 7] Pair Verify Step 2/2
>>> HomeKit: [Client 7] Found pairing with XXXXXXXX-YYYY-ZZZZ-AAAA-BBBBBBBBBBBB
>>> HomeKit: [Client 7] Verification successful, secure session established
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with XXXXXXXX-YYYY-ZZZZ-AAAA-BBBBBBBBBBBB
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: Got new client connection: 9
>>> HomeKit: [Client 7] Get Accessories
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 9] Pair Verify Step 1/2
!!! HomeKit: [Client 7] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 6] Get Accessories
!!! HomeKit: [Client 4] Error reading data from socket (code 104). Disconnecting
>>> HomeKit: [Client 7] Closing client connection
>>> HomeKit: [Client 4] Closing client connection
!!! HomeKit: [Client 9] Error reading data from socket (code 104). Disconnecting
!!! HomeKit: [Client 6] Error reading data from socket (code 104). Disconnecting
>>> HomeKit: [Client 9] Closing client connection
>>> HomeKit: [Client 6] Closing client connection
!!! HomeKit: [Client 5] Error reading data from socket (code 104). Disconnecting
>>> HomeKit: [Client 5] Closing client connection

bbfelts, 2019-04-28:
The issue I posted above only seems to happen on v.0.8.5 for me (I installed OTA).  I downloaded the v.0.8.4 and installed directly and it paired successfully right away.

closed 2019-05-26 by RavenSystem