RavenSystem/esp-homekit-devices #357: Select Characteristic

MarcioPlaza, 2019-04-28:
Good Morning,
Download the .bin files and run as per instructions, worked perfectly, however I'm trying to switch to Garage Door and I do not know when I do this? Sorry for the question maybe this is simple but as I'm starting up in development I'm having these difficulties of understanding if I have to compile main.c for main.bin or if it's some simple procedure I'm doing wrong. Thank you and thank you very much.

tohmc, 2019-04-28:
Do it with Eve App.
Ther you have your Option „Show Setup“
Under this you have your options like device Type and Board Type.

MarcioPlaza, 2019-04-28:
wow it worked did not know it was that easy.  Thank you so much, I earned my Sunday.

closed 2019-04-30 by RavenSystem