RavenSystem/esp-homekit-devices #358: Make Shelly2.5 rotate off/off-on/off-off/on-on/on with single external momentary switch

kittmark, 2019-04-28:
There are custom firmwares out there capable of making the Shelly2/2.5 behave in such a way as described in the title by just repeatedly pressing a single external momentary switch, could this mode (“rotating latching relay”) be implemented in RavenCore? So 2 groups of lamps can be controlled with just one external wall button.

1st press: relay1 on, relay2 off
2nd press: relay1 off, relay2 on
3rd press: relay1 on, relay2 on
4th press: relay1 off, relay2 off
...and so on back to the beginning

thanks

GPL71, 2019-04-29:
It seem you're asking this feature.
It would be very interesting!!

commutador finder ref.27.05/25
https://gfinder.findernet.com/public/attachments/27/ES/S27ES.pdf
sequenza finder

kittmark, 2019-04-29:
Right, I’m exactly asking for the shelly2/2.5 to behave like a Finder 27.05. 
Lots of people need it to make existing Finder 27.05 light setups smart.

RavenSystem, 2019-04-30:
I can code this.

GPL71, 2019-04-30:
wow!!! In Italy has a large use .....

kittmark, 2019-04-30:
Would be awesome...by the way much appreciation for your work, really great...going on paypal right now to show it a bit...

RavenSystem, 2019-04-30:
@kittmark Many thanks!!!

A little question: this mode must use a button and not a switch, right?

kittmark, 2019-04-30:
Yes a button (brief pulse).

RavenSystem, 2019-04-30:
@kittmark Upgrade to 0.8.6 and select external toggle 1 = 3, or external toggle 2 = 3 (or both if you want to use 2 buttons, but both will do the same).

GPL71, 2019-04-30:
@RavenSystem just added?
I can't believe!!!

kittmark, 2019-05-02:
Awesome! I’m going to try to find the time to test the new function today or tomorrow.

closed 2019-05-10 by RavenSystem

curla92, 2019-05-18:
hi josè i would to know if possible change the first lamp of the first click. For example if i have 2 external toggle i want that the first of them turns on the first lamp at the first click and the external toggle 2 turns on the second lamp at the first click. 
I hope it's clear. thank y