RavenSystem/esp-homekit-devices #359: Feature - External toggle

claire078, 2019-04-29:
Hi,
I use a normal thermostat to switch a Sonoff Basic to control heater. 
There is two external modes: button or wall switch. 
It would be good if would a third mode, in this mode always switches according the state of RX - Ground (closed or open). 
If thermostat on and if I manually switch off Sonoff then if thermostat changes state switch stays off.
I hope I could explain it. 😊

RavenSystem, 2019-04-30:
I really dont understand you. 😅

claire078, 2019-04-30:
Try to ask. 😉
Switch according to RX-Ground opened or closed. 
Don’t change state if there’s a short on RX-Ground.
If it is closed then ON if open then OFF.

RavenSystem, 2019-04-30:
And "board type = 4" and "external toggle 1 = 2" doesn't make sense?

claire078, 2019-04-30:
I use it now with this settings. 
What if thermostat OFF I turn sonoff ON then thermostat turn on..?

claire078, 2019-05-05:
Never mind. I’ll figure it out how could I use it as normal thermostat with sensor.

closed 2019-05-05 by claire078

claire078, 2019-05-12:
Is this in 0.8.10?
Thanks. 👍

claire078, 2019-05-12:
I tested it but external toggle mode 3 makes nothing at all. 😟
There is nothing about it in wiki.