RavenSystem/esp-homekit-devices #360: Power monitoring/capping for Shelly1pm and 2.5 in Eve

kittmark, 2019-04-30:
Hello, as we all know there is no power monitoring in Homekit, but said feature (monitoring and auto-off after a set threshold) could be implemented within the “Show setup” function in Eve. 
Also I think some people will end up using the PM versions of the shelly (1pm and 2.5) by default because of the added thermal protection, so why not exploit the power monitoring feature in RavenCore too. 
Thanks

RavenSystem, 2019-04-30:
I want to add auto off based on power monitoring. Do you know what esp-open-rtos libraries are used to measure power with Shellys?

kittmark, 2019-04-30:
I personally don’t know and know nothing about coding, but I’ve seen ESP8266 power monitoring stuff in Eve from the screenshots of a firmware called “Homekit Easy”, you could contact the author on facebook.

GPL71, 2019-04-30:
@kittmark The author sell its firmware..... they aren't for free

Shelly has an internal power meter... maybe in Tasmota Git we can find infos.

After exist PZME-004T that is a power meter and it is possible join it to an esp 8266.
Then I image you need code the part for PZME-004T and forward data and at same time set a limit in EVE. 
If you rise up limit then a memory wake up and by ESP8266 you can switch on a relay.
Then you can reset relay manually tapping on icon in HOME app if current turn under limit

RavenSystem, 2019-04-30:
Selling a non certified HomeKit firmware is illegal, and pay for it, too.

In other hand, I am going to looking for some info about PZME-004T to use. I am very interested in power meter to include it with Shelly2.5 and rolling shutters to avoid motor damages.

GPL71, 2019-04-30:
@RavenSystem Josè I agree .... it is illegal both things.

About PZME-004T it would be interesting measure adsorbed power in an electrical line and if you set a limit give priority to one or more loads switching on or off relays.
An Homebridge plugin manage PZME-004T and some github repo too.

RavenSystem, 2019-04-30:
I must investigate this when I get some free time.

GPL71, 2019-04-30:
@RavenSystem I'll forward some info about

greensouth, 2019-05-03:
Yeah! also some fine-tunning with shelly 2.5 + rolling shutter would be great, because the % is not as accurate as it should. I read someone who said something about adding some delay to the beginning of the movement.

We have to keep in mind that for the first couple of seconds the roller isn't moving the cover, it's straightening the clips (or the FLEJES para entendernos xDDDD) so it's not really moving. In fact, if I set 50%, visually it seems more 20-30%.

I remember that the official Shelly software, has obstruction detection and a few more options. I don't know if it's possible to ask them how to access the values, due to its 'open nature'.

Added to the power metering feature , would be a great step forward to implement the advances features of this device. :)

GPL71, 2019-05-03:
@Josè .... 
some info about power meter PZEM004T and ESP8266
https://github.com/olehs/PZEM004T/wiki/Connecting-PZEM-to-ESP8266
http://pdacontrolen.com/meter-pzem-004t-with-arduino-esp32-esp8266-python-raspberry-pi/
https://github.com/olehs/PZEM004T/wiki/Basic-Setup
https://sprut.ai/client/article/1111

RavenSystem, 2019-05-29:
This is the driver that I need to migrate to esp-open-rtos: https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xnrg_01_hlw8012.ino

mgoeller, 2019-09-09:
I think this is the correct driver for Shelly 2.5, right?
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xnrg_07_ade7953.ino

closed 2020-02-07 by stale[bot]

aleroot, 2020-02-18:
I don't understand why this feature request has been closed, since it could be useful to many ...

RavenSystem, 2020-02-18:
You can see this feature is in the roadmap: https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List

dcdomotics, 2023-01-22:
Hi, Is there a chance that pzem will be implemented?