RavenSystem/esp-homekit-devices #362: 4ch pro on 0.8.6

dwhp, 2019-05-01:
Hi. I just did an auto update on 4ch pro from 0.8.5 to 0.8.6. 

After reboot it did not appear in home/eve app, pressing physical buttons ok. 

So I  did a factory reset, default mode one was ok, however after changing to mode 4 again the issue came back. (Not responding in home/eve app however physical buttons ok.)

Is this just me?

Thanks

RavenSystem, 2019-05-01:
Bug confirmed.

RavenSystem, 2019-05-01:
Try version 0.8.7.

dwhp, 2019-05-02:
Working again now with 0.8.7. Did have to try auto firmware update twice as first time it got stuck in ?ota? .... anyway like I said, all good now. Thanks :)

Ps looking forward to HAA. I have a couple of different non-Sonoff sockets (like a S26) that I bought to experiment with.

closed 2019-05-02 by dwhp