RavenSystem/esp-homekit-devices #363: A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

appinn12, 2019-05-01:
D:\IOT Projects\Pythons\Scripts>esptool.py
esptool.py v2.6
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...

esptool.py v2.6 - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    version             Print esptool version

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32}, -c {auto,esp8266,esp32}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset}, -a {hard_reset,soft_reset,no_reset}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use
                        with care)

D:\IOT Projects\Pythons\Scripts>esptool.py -p COM3 erase_flash
esptool.py v2.6
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

[image]

RavenSystem, 2019-05-01:
Or COM3 is wrong, or your device is not in flash mode.

appinn12, 2019-05-01:
I tested on the window system  when we check device manager  that is my port name 

for flash mode when I connect the device to system , I press the device button 2-3 seconds and leave button after connecting

appinn12, 2019-05-01:
can you pls provide me steps if I am doing wrong because I face this issue many times while connecting 
[image]

RavenSystem, 2019-05-01:
Sorry, I don't use MS Windows.

appinn12, 2019-05-01:
do you have any idea for run this devices on the window system or any other way becuase I am stuck here in last 2 days , I used SONOFF devices connected with usb cable and follow your steps but when we try to connect ,the upper error shown. your help is really appreciates for me for solve this problems

RavenSystem, 2019-05-01:
Maybe Sonoff is not connected right. Try to swap rx and tx wires.

GPL71, 2019-05-01:
usually the problem is solved swapping tx and rx

closed 2019-05-26 by RavenSystem