RavenSystem/esp-homekit-devices #364: Join three .bin

kalety, 2019-05-01:
Hi, Jose!, I have several outlets tuya esp8266 based devices . I'm testing it, but flash this devices is very difficult... Is there any way to join the three bin's needed for Ravencores's flash in one bin alone?. This let flash by tuya-convert you firm without soldering, via OTA.

Thanks in advance

RavenSystem, 2019-05-03:
Yes, it is possible to join them and create a unique bin. But I understand that you are talking about otaboot firmware instead RavenCore. The problem here is that is you don't erase flash completely, you will have issues when you load firmware because some sector of flash are not erased. But if you want to try, I can give you an unified file.

kalety, 2019-05-04:
If were possible erase it completely, it works?. I'm trying change the tuya’s firm by your RavenCore, but this devices are very difficult to flash and found the GPIOs... Other alternatives firm, like tasmota   or ESPurna are possible flash by tuya convert, but I like yours...;)

RavenSystem, 2019-05-04:
Here is. Remember to unzip first:
[complete_otaboot.zip](https://github.com/RavenSystem/esp-homekit-devices/files/3144816/complete_otaboot.zip)

kalety, 2019-05-05:
Thanks,! I will test it and will tell you!

jorgemartins, 2019-05-06:
hey @kalety, I was looking into Tuya-convert method to better understand it, and looks like it doesn't "erase completely" the flash. as per [173 issue on tuya-convert repo](https://github.com/ct-Open-Source/tuya-convert/issues/173#issuecomment-483045820)
> flash2 only erases sectors 128-256 (0-0x80000), flash3 only erases 0-128 (0x80000-0x100000). If this is causing issues, I propose we do a full erase on flash3.

Although the developer shared [a build that would full erase](https://github.com/ct-Open-Source/tuya-convert/issues/173#issuecomment-486824011) is that what you'll use to flash with RavenCore joint bin?

thanks

kalety, 2019-05-06:
@jorgemartins Thanks for the info!!!!. @RavenSystem Other user test the ravencore firm flashing with soldering, no success... seems the GPIOs Ravencore's map don't match with this tuya devices. The GPIOs of this device are:
Maxcio EU Wifi Smart Socket W-DE004
GPIO1 = Button1
GPIO13 = Led1i
GPIO14 = Relay1

This is the amazon.es link...
[https://www.amazon.es/gp/product/B077G7R5GW/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1](url)

RavenSystem, 2019-05-06:
Of course RavenCore GPIOs don't match. Tis device will must use HAA firmware instead, but the idea of combined bin file to load LCM firmware (aka otaboot) using Tuya-Convert.

kalety, 2019-05-06:
Then... we will wait with impatience!!!
Thanks!

Alserser, 2019-05-10:
> Of course RavenCore GPIOs don't match. This device will must use HAA firmware instead, but the idea of combined bin file to load LCM firmware (aka otaboot) using Tuya-Convert.

I have built a version of Ravencore with an additional Board Type which maps the pins according to Maxcio EU Wifi Smart Socket W-DE004 device but, in order to flash it via Tuya-Convert I guess a change is needed in the build process (maybe I am wrong). I gave a look at Makefiles but I didn't come up with a solution. If you could shed some light on this I would appreciate it.

Thank you for all the work you are doing!

jorgemartins, 2019-05-10:
@Alserser have you tried using Tuya-Convert's build that fully erases the flash (Like it is required by RavenCore) ?
https://github.com/ct-Open-Source/tuya-convert/issues/173

Also, could you explain your process of re-mapping the GPIOs? hopefully the https://github.com/RavenSystem/esp-homekit-devices/issues/206#issue-401148373 will solve this, but in the meantime i'd like figure how I could do it myself (still clueless though), thanks in advance

RavenSystem, 2019-07-20:
No news about that. Closing...

closed 2019-07-20 by RavenSystem