RavenSystem/esp-homekit-devices #365: Make the Lock tile in Home long press like Security System

kittmark, 2019-05-02:
Nobody wants to open a lock by accident, especially if it is an open-only situation (and you have to walk to the gate to close it).

I’ve seen that to arm/disarm the aqara security system you have to long press the tile in Home.

Is it possible to have a “lock” (device with inching time and security notifications basically) with a tile that needs to be long-pressed like the security system type? It could even be a fake security system that actually just behaves like a lock type.

RavenSystem, 2019-05-02:
> Nobody wants to open a lock by accident, especially if it is an open-only situation (and you have to walk to the gate to close it).

Totally agree. I want to use lock and garage door in my new house, and I don't want to worry about that.
 
> Is it possible to have a “lock” (device with inching time and security notifications basically) with a tile that needs to be long-pressed like the security system type?

I can not do that, because it is managed by Apple. IMO, Apple should do something with garage doors, doors, and lockers.

> It could even be a fake security system that actually just behaves like a lock type.

In garage door and lock, I can attach the dummy switch (if it is enabled) to the lock. So, when you press lock button in your iPhone/iPad, it will work only when dummy switch (you can change its name to "Lock secured", for example) is off.

claire078, 2019-05-02:
Stateless switch could help?
I interested in as well because I use SV to control alarm system. 
And I can’t wait to use SV as a motion alarm device.

lizzus, 2019-05-07:
Can you add a timer for relay on lock device? I try to explain: Inching is a good solution for open electrical door (like door with a 12v electric door lock), but if for example we want adapt your device to a door like like "danalock" (not danalock because is native homekit) we have problem that relay (and lock) will be open too much time (if we set inching to 0). If we set inching to (for example) 4 seconds, that is time needed to mechanism to open lock, after this time status automatically go to "closed", but is not true because our door lock is open.

pilot1981, 2019-05-07:
Hi,
I also suggest to modify this type of device to use also in a more secure garage door scenario:

1) I use inching time of 4 second on home app
2) it unlock magnetic block + delay 1 or 2 seconds
3) close contact of garage door to open it

I suppose this could be done using n.2 sonoff basic?! 

I could be more secure and useful to more people

kittmark, 2019-05-07:
About what lizzus asks, I’m gonna propose my idea of how it could work using a combination of two devices (shelly to open the 12v lock and door sensor), because clearly currently the “locked/unlocked” status makes no sense when used with brief inching and with doors that need to be opened and closed again to re-engage the lock.

So it should work this way
- real inching: 0.6s (to actually open the 12v lock
- fake inching: 0 (“Lock” icon appears unlocked indefinitely in Home)
- when the door sensor is opened and then closed again, the “Lock” icon in Home becomes locked again

Not sure if this can be done in Homekit Architect combining 2 device or in RavenCore using the GPIO or something...

Currently I use the Lock icon just as button but its status and notifications are false and misleading obviously...a door sensor is need to really know if it gets reengaged..

lizzus, 2019-05-07:
@kittmark you suggest a dual device using a gpio for sensor?

kittmark, 2019-05-07:
I’m not even sure about what I suggest :D I’m
not very knowledgeable of coding. But I suggested that kind of logic: 
1st device: relay (shelly, etc.) to activate the 12V lock
2nd device: some kind of door sensor to keep track of the real locked/unlocked status of the lock, maybe via GPIO or something

Such 2 devices should be combined in a single “Lock” (type 13) device in Homekit.

lizzus, 2019-05-07:
I mean 1 physical device and 2 "virtual" device...

kittmark, 2019-05-07:
Now that you make me think about it, if José gave us a Lock+Dummy_switch double device, we could link the status of the dummy switch to a door sensor via homekit automation, José would then just need to code the device in a way that makes the Lock stay unlocked until certain things happen to the dummy switch...like
(user homekit automation): if door sensor GETS closed while it was open —> turn on dummy switch
(josé coded automation): if dummy switch is turned on —> make the associated lock look locked again
(josé coded automation): if lock is unlocked —> turn dummy switch off, keep the lock look unlocked even if you actually sent a brief pulse (0.6s inching)

this way we can track the full cycle
0) lock entity locked, actual lock locked, actual door closed, dummy switch off
1) lock entity unlocked, actual lock unlocked, actual door closed, dummy switch off
2) lock entity unlocked, actual lock unlocked, actual door open, dummy switch on
3/0) actual door closed, dummy switch off, lock entity locked, actual lock locked

José should just logically connect the dummy switch and the lock entity, we users should connect the door sensor to the dummy switch via homekit automation.

RavenSystem, 2019-05-26:
Included in TODO list.

closed 2019-05-26 by RavenSystem