RavenSystem/esp-homekit-devices #366: Critical !!! Wi-Fi resets after power failure

St0nehead, 2019-05-03:
Today, after power failure, all my RavenCore devices has been lost their WiFi settings.
i'm reconfigured 10+ devices, to make all of them work again.
FirmWare Version: 0.8.7

peros550, 2019-05-03:
I thing after a power failure I lost one device running custom firmware but based on the framework (esp-homeKit & esp-open-rtos) of this repo.

RavenSystem, 2019-05-03:
Personally ,I have never had a similar issue.

St0nehead, 2019-05-04:
Second power failure and same issue — all my Sonoff devices has been resetted. Please fix!

closed 2019-05-26 by RavenSystem

St0nehead, 2019-07-14:
Still the same annoying bug — after each power failure, WiFi resets and Sonoff switches to AP mode (with 192.168.x.x ip address).
Please fix! Really annoying problem!!! 🙏

peros550, 2019-07-14:
@RavenSystem are you planning to use esp-wifi-config latest version in your code?
That code uses a more sophisticated approach with the AP creation.

RavenSystem, 2019-07-20:
@peros550 no, because it enters into AP mode when wifi is lost, creating a security risk. I prefer my approach, that keep trying yo reconnect to known wifi and reannounces mDNS when success.

RavenSystem, 2019-07-20:
@St0nehead sorry, but I am not be able to reproduce your issue.

St0nehead, 2019-07-20:
1) Turn off your WiFi router. After 10-30 minutes, turn it on again.
When Sonoff lost WiFi connection, it turns to AP mode.
2) Found temporary solution:
After power failure, i wait router boot up.
Then interrupt power on sonoff devices (but not WiFi router). Sonoff reboots and connects to WiFi.

peros550, 2019-07-20:
> @peros550 no, because it enters into AP mode when wifi is lost, creating a security risk. I prefer my approach, that keep trying yo reconnect to known wifi and reannounces mDNS when success.

Hi Jose!

Why turning into AP mode is a security risk?

Kienz, 2020-02-12:
@RavenSystem What happens to installed HAA devices if the WiFi router is powered off? Found these two sentences in the documentation (https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode):

 _"If the configured WiFi network is unavailable then connect directly to a WiFi AP via the generated SSID e.g. HAA-XXXXXX (where XXXXXX are the last six characters f its Ethernet MAC address)."_

_"If the WiFi network becomes unreachable, the device will continually retry its connection until it becomes available again."_

If it creates a WiFi AP the device could be taken over by any person in WiFi range of the IoT device. Correct? If yes, is this a security risk?

RavenSystem, 2020-02-13:
> If it creates a WiFi AP the device could be taken over by any person in WiFi range of the IoT device. Correct? If yes, is this a security risk?

Correct, but AP is created only in setup mode when WiFi is unreachable, never in normal mode.