RavenSystem/esp-homekit-devices #368: Device 14

PNM96, 2019-05-03:
Hello! I have a light bulb that, I can't get to work with device 14. It's based on esp8266 but managed by MY9291 LED driver. 
I think the gpio are different ... do you have some suggestions?
The name of this bulb is AiLigjt. 
Thank you

lizzus, 2019-05-03:
did you try setting correct pin?

PNM96, 2019-05-03:
Yes! I have tried with all pins. 
I also posted the question on our group 😁@lizzus

RavenSystem, 2019-05-03:
MY9291 LED driver is not supported. Only PWM MagicHome based.

PNM96, 2019-05-03:
Thank you very much! 
In the future can be the possibility that it is supported?

P.S. Thank you for your work and for your support

RavenSystem, 2019-05-03:
I want to add other RGBW chips, selected under "board type" user option.

PNM96, 2019-05-03:
Great

closed 2019-05-10 by RavenSystem