RavenSystem/esp-homekit-devices #369: Rotary valve and Sonoff dual

i3laze, 2019-05-03:
I’m thinking of buying a 220V rotary water valve.
It differs from solenoid valves as it uses two line wires for control (1st to rotate ball to open position, 2nd to close), thus requiring Sonoff Dual in interlock mode.
I guess, no inch or inch up to 25-30sec would require.

Any way to support such valve as 1 device natively?

RavenSystem, 2019-05-03:
It must be a new device type, with new functions.

RavenSystem, 2019-05-19:
You will can do it with HAA when it supports water valves.

closed 2019-05-19 by RavenSystem

i3laze, 2019-05-20:
Actually, RavenCore with dummy switch to store state and bunch of HomeKit rules would do the trick too..)

i3laze, 2019-05-26:
So.. I've finally got the shipment, including Sonoff Dual R2.
Here's my nifty setup:

It works absolutely great in Device 12 mode (Window Covertings).. with several hickups:

1.  Although Button worked at Device 2 (Dual Switch), it does nothing for Device 12.. (I've tryied board types 1,2,4).
2.  I'd still prefer it as Water Valve type :) can you introduce one for RavenCore some day?