RavenSystem/esp-homekit-devices #370: OTA update problem

greensouth, 2019-05-03:
Hi José! I'm back with a new issue.

A few mins ago tried to update the MagicHome controller to the latest version, and after a few minutes the LEDs flashed and I thought it has finished.

But now the device appears as not accessible. Tried to unplug... Nothing. Tried to remove from Homekit and re add again and the device doesn't shows.

Neither does in my wifi settings although I see it connected to my access point.

Any clue not involving resoldering and reflashing?

BTW I hope you received my small contribution :P

RavenSystem, 2019-05-03:
Never remove an accessory from HomeKit if accessory is not accessible.

Power on it again, and wait 5 minutes. The try to find its wifi AP RavenCore-XXXX. If you don't see it, check your wifi AP and search your device.

There are 2 possible scenarios:
1.- OTA firmware is loaded: you must wait until firmware is updated.
2.- RavenCore is loaded: When you boot your device, you can connect GND with GPIO0 10 seconds to perform a factory reset.

A device with RavenCore and OTA firmwares never should be reflashed.

RavenSystem, 2019-05-03:
> BTW I hope you received my small contribution :P

Yes! Many thanks!!

closed 2019-05-10 by RavenSystem