RavenSystem/esp-homekit-devices #371: QR CODE pairing enabled for RavenCore firmware

GPL71, 2019-05-05:
Hi to all of you ....
@RavenSystem Josè has enabled the qr code pairing system in the Ravencore firmware.
Now we can use the label below to join our devices to Home APP
[image]

RavenSystem, 2019-05-26:
@GPL71 Can I add this image to Wiki?

GPL71, 2019-05-29:
@RavenSystem ..... It's generated using @MaximKulkin routine

RavenSystem, 2019-05-29:
@GPL71 I know it, but you are the person who has used your time to generate it.

GPL71, 2019-05-29:
@RavenSystem  Thank you for your kind consideration ... I'll hope you'll use it in the WiKi

mohamedkeven, 2019-05-31:
code not working

RavenSystem, 2019-07-20:
@GPL71 Please, can you create a new qrcode for RC 1.0?

GPL71, 2019-07-20:
Yes, surely @RavenSystem

GPL71, 2019-07-20:
[image]

This is a Maxim Kulkin routine

closed 2019-11-07 by RavenSystem