RavenSystem/esp-homekit-devices #375: FREE Flasher tool for RavenCore firmware

GPL71, 2019-05-06:
Hello to all of you ...
@Lizzus and I are the admin of the Italian facebook group and forum dedicated to HomeKit .

Our Group **Homekit Italia Group** is reachable by following links:
[https://www.facebook.com/groups/homekititalia](url)
[https://www.homekit.group](url)
[https://www.instagram.com/homekititalia/](url)
[https://twitter.com/homekititalia1(url)

We're @RavenSystem fans and followers too and we wants thanks for its great work.

 **Gianluca Scarlino**, an user of our Group has compiled a free flasher tool for our Italian community.
According to Gianluca we want share the tool with all RavenCore user. 
The tool is a contribute for the great work made by Josè.
The tool is setted for RavenCore, but it can be used for other firmware too.

Gianluca Scarlino works hardly to improve features and provide upgrades.
Below link to download the tool and instructions.
https://www.homekititalia.group/forum/viewtopic.php?f=40&t=46

Thank you very much to Josè from @Lizzus and me for allow us publish this message and the links of free Italian Group.
Thank you very much to Gianluca Scarlino for its free wonderful work

For any issue and suggestion open an issue in the forum

Thank you
@GPL71 @Lizzus

lizzus, 2019-05-06:
This program is for windows O.S. 
Has, among its various features, two buttons: one to flash the OTA firmware and one to flash the latest version of the Ravencore firmware (when the system starts it checks the latest version available. If. a new version is detected the program updates the local repository)

[image]
Support forum (in italian language, but english accepted) avaible here: https://www.homekititalia.group/forum/viewforum.php?f=40

St0nehead, 2019-05-07:
Homekit flasher for Windows? What? 🤦🏻‍♂️

mafyata, 2019-05-07:
Why not? I flashed my sonoff devices with my windows laptop.

GPL71, 2019-05-07:
> Homekit flasher for Windows? What? 🤦🏻‍♂️

There are a lot of windows users ....
However we hope release for osx users

greensouth, 2019-05-07:
Well a GUI is always welcomed but I do it with my PI via terminal with my Mac hahaha

mafyata, 2019-05-07:
Excuse me , but I am not rich like you to own a macbook. This program is useful for me (windows user). Thanks to homekititalia guys.

RavenSystem, 2019-05-07:
I personally don't use MS Windows, but I think that many people that has iPhone/ipad... uses it. Apple supports MS Windows for iTunes, Apple Music, iCloud, Mail, and a lot of Apple features.

If anyone doesn't want to use this flasher tool, good, but let other people to do what they want. It is a tool that a person has created and shared for free.

And other thing: Maybe some people prefer to flash from a cheaper Windows machine instead from an iMac Pro with 12 cores because some FTDI adapters can be dangerous in some cases and can burn up your USB port... 

Using this software is your choice, but doing destructive critics, no (In my repo, that is not allowed -> Why? My repo, my rules).

tohmc, 2019-05-15:
Wotka Great!

Thank you for your work!

lizzus, 2019-05-15:
I'm very happy. If you like it, you can click on "like button" on app
[image]

peros550, 2019-05-16:
Well done for creating and providing this tool to the community. May I ask two questions. 

Is the tool able to download latest firmware files from github?

What the “Extract Mac” button is used for ?

lizzus, 2019-05-16:
I answer:
1. Yes tool download automatically latest version of Ravencore, but not from github. It download from our repository
2. The extract Mac is only for extract MAC address of a device.

closed 2019-05-26 by RavenSystem

cyberhusky69, 2019-09-01:
Hello
I'm new to all this homekit possibilities and fascinated by it. 
Can I run the Ravencore flasher tool in Virtualbox under Windows 10 on my iMac?
Or is it better to use a PC?

Thx

lizzus, 2019-09-01:
Yes, you can