RavenSystem/esp-homekit-devices #377: Lost Wifi connection after automatic channel switch by router

marcsowen, 2019-05-07:
Hi,

my internet router as a feature, where it periodically scans all Wifi channels for interference and occasionally decides to switch channel to a less crowded one. At this point my Shelly 1 often times looses connection and cannot reestablish a connection to the new channel.

Example from my router logs:

```
02:00:05 - Router - Connection to Shelly 1 established. 72 Mbit/s. IP address: 192.168.178.42
14:00:54 - Router - Start of channel scanning...
14:01:06 - Router - Channel switch from channel 6 to channel 1.
14:01:06 - Router - Connection to Shelly 1 lost. IP address: 192.168.178.42
14:01:07 - Repeater1 - Connection to Shelly 1 established. 54 Mbit/s. IP address: -
14:19:59 - Repeater1 - Connection to Shelly 1 lost. IP address: -
```

After that the Shelly 1 is offline for the next couple of hours. Sometimes the shelly reestablishes a connection again, but it might take up to 6-12 hours and it's completely unclear what triggers it. I'm not sure whether the link quality between the Shelly and the Repeater1 is sufficient to even make communication possible. (Router and repeater are linked via Ethernet cable).

Sometimes the Shelly reestablishes a connection to the router again after channel switch without a problem. But it seem to fail more often than it succeeds.

I think the Shelly should do the following:

- Always try to connect to the Wifi channel with the best reception. Do not stick to the current channel.
- If there is no IP address, the DHCP client should more or less aggressively try to get one, since the current Wifi channel might not allow flawless communication.

What do you think? Maybe my analysis is flawed. Do you have any hints on how to debug this situation better?

Thanks for your help!

RavenSystem, 2019-05-07:
Hello @marcsowen ,

This is an esp-open-rtos issue, but I think problem is that wifi signal is not sufficient. I see that when Shelly1 connects to your Repeater1, link is at 54Mbit/s, that cases with 802.11g. This is bad, because it should connect using 802.11n, as your router shows, with 72Mbit/s. Shelly1, and this kind of devices, have a minimal printed antenna with a low power signal.

My router changes wifi channel when is necessary, and my devices reconnect very fast (a couple of seconds).

If you want to debug more, you can connect your Shelly1 to your computer using your FTDI adapter and see the console log (https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs). You will see when device disconnects from wifi and reconnects.

marcsowen, 2019-05-15:
Thanks for your help. Is it possible to get the log over Wifi? My Shelly is installed behind my wall switch and connected to 230V so I cannot just plug in the serial interface. I tried "nc -kulnw0 45678" but without any luck. Maybe I didn't wait long enough...

RavenSystem, 2019-05-15:
It is not possible. "nc -kulnw0 45678" only works when you are performing an OTA update. I want to include that option to collect general logs over wifi, but actually it is not implemented.

Can you move your repeater near of Shelly1 to check if it connects with a speed upper than 54Mbit/s? And can you check if other devices like your iPhone connects to your repeater with a high speed at 2.4GHz band?

RavenSystem, 2019-07-03:
Closed due to inactivity. Reopen if you need.

closed 2019-07-03 by RavenSystem