RavenSystem/esp-homekit-devices #380: 2x Sonoff Basic > Problem with the same Homekit Code?

kugelkopf123, 2019-05-08:
Hey! First of all: Thanks!

Today i flashed 2 Sonoff basic switches from a friend via OTA Method as it is explained in the wiki. 
No Problem with that. I could add them in his Home App individual. But if you power up both devices no one works. 
If i plug off both and connect only one of them it works perfect. 

Perhaps its a Problem with the Homekit Code? Because on both devices its the same code. 
Perhaps it would be nice if you can change the code to a personal one within the Wifi Screen. That would be nice.... 

I don't know if this is really the problem but... yeah. Thats what happend.

GPL71, 2019-05-09:
don't worry for pairing code 
I have 18 shelly working with the same pairing code.

kugelkopf123, 2019-05-09:
> don't worry for pairing code
> I have 18 shelly working with the same pairing code.

Hm. Idk. Its strange that its not working when both sonoff basics are online. They have different IPs. So that cant be the problem. I built 3 new Firmwares with different Code now. I will have a try over the next days and going to tell you what happend.

RavenSystem, 2019-05-11:
@kugelkopf123 I have many devices, all with same code and all works fine. And remember that changing code is not allowed: https://github.com/RavenSystem/esp-homekit-devices/blob/master/LICENSE

closed 2019-05-11 by RavenSystem