RavenSystem/esp-homekit-devices #382: Possible error in external toggle (device 12)

viccur, 2019-05-10:
Hello.

Thank you for this great job.
I use it in a lot of devices.
Recently I've installed a shelly 2,5 in a roller Shutter.
All works OK except external toggle. 
I have a roller switch and I try to configure it in "External toggle = 2". After reboot the efect is the same as "External toggle = 1".It acts as a push button.
The version is 0.8.7
I have another shelly 2 in other roller Shutter with version 0.7.2 and all works OK.
¿Am I make somthing wrong or is a possible bug?

Best regards.

RavenSystem, 2019-05-10:
“External toggle 1 = 2” must work. Have you reboot your device after setting it?

viccur, 2019-05-10:
Yes. Several times. And I tested it in three modes:
0: it don't do nothing.
1: when switch to down, it don't do nothing. When switch to stop position, it goes down to the end. If you repeat with the switch to up, it's the same. Only goes up when you return to stop position.
2: exactly the same as 1.
The settins are:
Actual Dev Type: covering
Board Type: 5
Cover Type: 0
Device Type: 12
Dummy Switch: Off
External Toggle 1: 2
Firmware Update: Off
Inching time DM: 0
Init State DM: 0
Log output: 1
Reboot: Off
Reverse SW1: Off
Reverse SW2: Off
Time Close: 19
Time Open: 20
Wifi IP Addr: 192.168 .....
Wifi Reset: Off

RavenSystem, 2019-05-10:
I will check it.

RavenSystem, 2019-05-11:
Try v0.8.8.

viccur, 2019-05-11:
Hello.
I has updated to the 0.8.10 and now the function has changed.
External toggle = 1: when switch to down, it starts to down. If switch to stop, it keeps going down. If switch to down another time, it stops. The same for the go up function.
This function appears good. It is a normal for push button or tactile touch. 
External toggle = 2: when switch to down, it don't do nothing. When switch to stop it starts to down.  It's no possible to stop without switch to up.
Today I can't do more test.
I will do it tomorrow.
Thank you so much for your job.

RavenSystem, 2019-05-11:
It is working in the opposite way; fix is very simple. Maybe Shelly2.5 is different than Shelly2. Can you update your Shelly2 to 0.8.10? I think problem is with Shelly2.5.

viccur, 2019-05-11:
Shelly 2 updated to 0.8.10 and works OK.
Tomorrow I will test it better, but is possible that this is the problem.
Thank you.

viccur, 2019-05-12:
Hello.
Confirmed. 
External toggle = 2: when switch to down, it don't do nothing. When switch to stop it starts to down. If switch do down another time, it stop. When you switch to stop it continues to down. 
Same when you try to up it.
How you said, apears to work in the opposite way.
Thank you.

mmiglier, 2019-09-01:
Hi, thanks again for your work. The same is happening to me. My existing switch is a temporized one. With shelly 2.5, ravencore 1.1.0, device 12 and external switch 1 = 2, is working in the opposite way:

- down switch set to on: shutter remains stopped
- down switch set to off: shutter starts closing
- up switch set to on: shutter stops
- up switch set to off: shutter starts opening

I tried enabling “reverse sw1” and “reverse sw2” to 1 but it was messing up with the motor and I immediately reverted the setup.

RavenSystem, 2019-09-01:
Sorry, I forgot this.

mmiglier, 2019-10-06:
Hi, I read you released the latest raven core release. Does it mean this bug won’t be ever fixed? I didn’t give HAA a try yet, is this use case already available and fixed with HAA? Thanks

RavenSystem, 2019-10-06:
Rolling shutters are not supported by HAA yet. It will be available ASAP, with this issue fixed.

closed 2019-11-07 by RavenSystem