RavenSystem/esp-homekit-devices #383: Garage Door - open with home app doesn't work properly

jat80, 2019-05-13:
Hi there,

I've wired everything up to my Sonoff SV using board type 1 with a NO sensor for the required Closed door sensor to GPIO 14.

If I use my own garage door remote TO OPEN the door it works as required in regards to the sensor working etc BUT if I use Siri or home app to open it, it opens the garage door but then stops the opening after 3 seconds or so and the home app still sayings garage door opening?

This only happens if I use the home app, Siri or eve app to open the door.

Have I configured something wrong?

[image]
[image]
[image]
[image]

greggsymington, 2019-05-14:
I have the same issue. This has been plaguing me with one of my two doors.

"Left" door works great, "Right" door will start doing it and even after deleting, reflashing and re-add to Home it lingers. Please see video HERE.

jat80, 2019-05-14:
> I have the same issue. This has been plaguing me with one of my two doors.
> 
> "Left" door works great, "Right" door will start doing it and even after deleting, reflashing and re-add to Home it lingers. Please see video [HERE](https://youtu.be/OTDG9QGhb3c).

Had a look at the video and this is what happens with mine but I don't use the button on the SV but use the HomeKit app to open it.   Can anyone help with this?  Do I have a faulty Sonoff SV?

pandarek, 2019-05-14:
Ty set „Has stop” to off.

jat80, 2019-05-17:
Ok, it seems to be working now.  I turned off the "has stop" but that didn't change anything.  I do recall trying this initially by turning it on when it initially didn't work.

What did seem to work though was changing the Reverse SW1 switch and everything worked after that regardless if I use Siri, the home app, or using the actually garage door remote control.

Really stoked it's all working now.  I'll close this now.  Thanks.

closed 2019-05-17 by jat80

skanico, 2019-06-07:
I also had to turn off "has stop" (although my button effectively stops the door) otherwise I got the same behavior (the movement stops a few seconds after each maneuver). I don't know what is the effect of "has stop" and if this can be problematic.